apache/plc4x #954: [Bug] [plc4go] [eip] AB EthernetIP connection stuck

charlyjna, 2023-05-18:
### What happened?

Hi everybody,

I am a new user of plc4x, and I wrote a very simple program based on example to connect to an Allen Bradley PLC using EthernetIP protocol.  It seems to be connected, but waiting for the 'connectionRequestChannel' to respond forever:

`// Wait for the driver to connect (or not)`
`fmt.Println("Waiting for the driver to connect...")`
`connectionResult := <-connectionRequestChanel`

Here is a screenshot of the logs:
[image]

Is it an Allen Bradley issue or a driver issue?

Thanks.
Charly

### Version

latest

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-05-18:
Are you using the latest SNAPSHOT version, or a released version? I'm asking, because in the current SNAPSHOT version we changed the EIP driver quite a bit explicitly in order to support AB PLCs.

chrisdutz, 2023-05-18:
But just reading that it's about PLC4Go ... the GO drivers are always a bit behind as most of the development happens on the Java versions. Unfortunately currently none of the Go-folks have a working EIP device, but I'm hopefull, that I'll be able to get my AB CompactLogix PLC working next week. However I probably would work on smoothing the edges of the Java version first.

charlyjna, 2023-05-18:
I am using the SNAPSHOT version: plc4go@v0.0.0-20230517143632-ef1c459fb118. I executed `go get -u` before my last try.

chrisdutz, 2023-05-18:
Yeah ... I guess that driver needs a lot of fine-tuning as the updates mainly happened to the Java version and they might have broken some of the Go stuff ... would you be willing to help with getting the go version in shape?

charlyjna, 2023-05-18:
I think I can help as I am trying to work with PLC4GO for an internal project. So if I can be helpful, sure!

chrisdutz, 2023-05-18:
I guess if you could give the Java version a try ... this should be the base-line version of the go version. If that works, I guess porting the Logix specialities would be the next step. Sorry that I can't help any more right now ... my Logix isn't configured yet and I still need some parts that will be delivered over the weekend also I'll be traveling till Sunday, so I won't be much help till then ... but till then ... if you could check the Java version and use a little WireShark ... should help find out what's different in the Go version.

chrisdutz, 2023-05-18:
Prior to b the current snapshot, the eip driver only used basic eip... The new version uses AB specific CIP objects... But it's less well rested.... That was why I bought the hard and software for my own setup. But I guess we need to put a bit of work into it

hutcheb, 2023-05-18:
The 0.11 SNAPSHOT has a pre-configured variant of the eip driver
specifically for the logix family.

https://plc4x.apache.org/users/protocols/logix.html

I believe this is only in the Java version though.

I would be very interested in seeing the pcap files from wireshark for both
the successful connection using 0.9.1 as well as a capture using the logix
driver on the 0.11 Java version.

Ben

charlyjna, 2023-05-19:
This morning I tried again using Java 0.9.1 eip driver, and 0.11.0-SNAPSHOT logix driver. Unfortunately, it failed for 0.11 with an exception. I attached wireshark results and console output.
I also attached the pcap file of eip Go driver which is, I think, waiting for an answer forever.
[wireshark-eip-go-java.zip](https://github.com/apache/plc4x/files/11516771/wireshark-eip-go-java.zip)

hutcheb, 2023-05-19:
Taking a look at the pcaps, 

The 0.9.1 looks like it is it reading 4 variables N40[17], N40[0], N40[16] and N42[67].

The 0.11 version is trying to read N40[17], N40[0].8, N40[16] and N42[67]. (Note the N40[0].8 address change, which is the problem as it can't find that address) PLC4X should definitely handle the response a little better though.

The 0.11 Go version is probably a write off, it really hasn't been tested at all.

If anyone else is looking at this, then changing the ENIP protocol's TCP port in Wireshark to 10123 allows the messages to be decoded.

hutcheb, 2023-05-19:
> I used the same variables for 0.9.1 and 0.11. And I used %N40[0]:8 for
> 0.9.1, then N40[0]:8 for 0.11 which does not like it.
>
> But, it works very nice if I use N40[0]. I do not now how to translate
> N40[0]:8 - N40[0]:WORD[8] or N40[0]:UINT[8] are not accepted by the 'logix'
> driver.
>
> Unfortunately, I am working with Go version and it is not possible to
> establish a connection. Is there anything else I can do for helping in
> solving this issue?

The addressing in the logix driver when addressing bits of words is
N40[0].8 (A period in stead of a colon).

Having a go at copying the implementation of the Java version into Go would
be a very good start. If you are unable or unwilling to do it, then you may
need to find someone with a shared interest in fixing it.

charlyjna, 2023-05-19:
FYI, I tried also with N40[0].8 but I still have the error using 0.11 Java drivers.

For Go drivers, I will try to work on it as much as I can.

Thanks for the help.

chrisdutz, 2023-05-24:
Today I got my CompactLogix working with a real program and ran into issues with PLC4J EIP driver ... however I adjusted our mspec for EIP and with these changes I was successfully able to read and write to my controller. 

Could you please check again with the latest version (SNAPSHOT)?

chrisdutz, 2023-05-25:
If I have a look at how we expect addresses to look  ... there probably should be something more ... as it's "{tag}:{datatype}:{elementNumber}"

charlyjna, 2023-05-25:
I tried:
- %N40[0] >    OK
- %N40[0].8 > Exception occured: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.eip.base.tag.EipTag.getTag()" because "plcField" is null
- %N40[0]:UINT > Exception occured: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.eip.base.tag.EipTag.getTag()" because "plcField" is null
- %N40[0]:UINT:8 > Exception occured: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.eip.base.tag.EipTag.getTag()" because "plcField" is null
- %N40[0]:8 > Exception explained in my previous message

Using 0.9.1 driver, this address %N40[0]:8 works fine.

chrisdutz, 2023-05-25:
I noticed that with the latest SNAPSHOT we weren't parsing adresses at all .. this caused problems when trying to write. So I changed that behavior.

charlyjna, 2023-06-20:
I made other tried using Java driver, and I have some issues I do not understand.
First of all, I can connect to my ControlLogix using logix driver and eip driver, but:
- Logix: I cannot read arrays like 'N40[0]:INT:16'
- Eip: I cannot read anything because I get an error explaining that the PLC is not using BIG ENDIAN format

Are you able to read array with your ControlLogix?

chrisdutz, 2023-06-21:
The logix driver is currently very new and still being fine tuned. If you find any issues like this with it. Please open separate issues.

chrisdutz, 2023-06-21:
But with the endianess, you can explicitly tell the driver to use little endianess by adding a connection parameter to the connection string. Currently on my phone, so challenging to look it up, but I'm sure others will be able to help.

sruehl, 2023-06-21:
I think it is just `bigEndian=false` as query param

chrisdutz, 2023-06-21:
Could you please create a network capture with the logix controller and add it to a new issue? I will try to have a look at it then.

charlyjna, 2023-06-21:
I just posted a new issue. Thank you very much for your help.

chrisdutz, 2023-07-21:
So? Is this issue closed? what's the status on it?

chrisdutz, 2023-07-25:
Aaaahh ... yeah ... sorry for that ... well I guess as long as nobody's willing to port the changes, it will be challenging. I could try to get the old version of the EIP driver working again (that would not have direct support for AB devices). 

It would however be cool, if you could test the Java version, if that would work as you expect it too as that would be the state, that we would port to Go.

Unfortunately I know that my previous employer would directly benefit from me working a PLC4Go EIP Driver able to talk to AB devices and I am not too eager to give him that for free.

I'm happy to help you or anyone willing to work on it.

charlyjna, 2023-08-01:
I am busy on other stuffs for now, but I will try Java driver very soon and let you know.
Because using PLC4X is very useful for me, and I think it is a very nice project, I would be happy to work on the Go version each time I can. But how? I need to learn current source code deeper first...

chrisdutz, 2023-08-01:
If you are really interested, I would reccomend to ask @sruehl if he'd have some time to walk you through the important parts (He's the grand master of the Go part ;-) )

chrisdutz, 2024-07-26:
So what's the update on this? Did the Java version work? Do you have time and are willing to help with PLC4Go?

charlyjna, 2024-07-31:
> So what's the update on this? Did the Java version work? Do you have time and are willing to help with PLC4Go?

Unfortunately, I was not able to work on fixing this issue as I had to work on my project so I used this driver: https://github.com/danomagnum/gologix. This one works very very well.

Anyway, I am still interested in working with PLC4X (Go), and I want to help for improving. I should be able to work on it for helping in the next few weeks.

chrisdutz, 2026-08-16:
So ... should we close this issue or will you be able to help addressing it?

I'll close it the next time I come around and don't hear anything.

closed 2026-08-18 by sruehl