apache/plc4x #959: [Feature Request]: UDT Examples

alison-gravley, 2023-05-22:
### What would you like to happen?

PLC/Protocols:
Not sure which one below is for AB PLCs, I think EIP? Specifically Logix and Compact Logix PLCs. Also Siemens (S300, 1200, and 1500) and probably OPC UA. All are over Ethernet connections. The table [here](https://plc4x.apache.org/users/protocols/index.html) seems to suggest that EIP is not supported? 

I would like to have a POJO for UDTs. I tried to find an example in your repo and it looks like the OPM requires the tag address to be hardcoded? We would have multiple tags of the same type. Potentially hundreds of the same type. This is the first time I have used this library and I am pretty lost. We are using Java with microservices (Spring Boot) and we would like to use open source libraries if possible.

I used [PLCIO](https://www.ctiplcio.com/), which is paid, for a few years at my previous company, and it was pretty nice to use, for C at least. One of the nice utilities it had was a discovery program that scans all of the tags in the PLC and then it could spit out C style structs for any UDTs, with a special conversion string you can pass once you get the plc read part of your program. I think it can work without the conversion string, but their manual doesn't mention it.

What I am trying to accomplish is this:
Create POJOs of PLC structs/UDTs that are commonly used in our programs. These may be fairly nested. Similar functionality to jackson xml/json.

Then, we would have our list of tags to read via polling, or subscription if supported, using the root tag name in the case of a struct, or the normal types such as BOOL, DINT, STRING, etc. We would like the UDTs to map to a registered POJO, or however you would need to do it.

Example:
POJO without nested UDTs. 
```
package com.mycompany.devicecommunication.plc4x.UDT;
import lombok.Data;

@Data
public class HoldStatusRequestUDT {
    //Do I use PlcDINT, PlcBOOL, etc  type or just an int, log, etc?
    //Probably need some sort of Alias annotation, or make the field name exactly the same as the PLC.
    private boolean TRIGGER;  
    private int sequenceId;
    private String barcode; 
}
```

List of tags of type "HoldStatusRequestUDT" that we would read from
SCAN100_REQ
SCAN101_REQ
SCAN102_REQ

[image]

and then I would expect updates for each tag and when we read, we would be able to map those values into the POJO.

Please let me know if this is already possible and maybe an example or two of how to do it. Thank you!

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [X] S7

chrisdutz, 2023-05-22:
Hi,

well actually you're joining the party to the exact right time ;-) ... Today I finally got the last parts delivered for setting up my CompactLogix controller ... we recently got quite a major refactoring of the existing EIP driver. However I wasn't able to actvely test it ... till now. 

So it is possible to read the tags and datatype information from the PLC and to read and write these UDT (User Defined Types ... for the others). 

We have somerhing similar in the Beckhoff ADS driver ... however the default is returning a map of strongly typed PlcValue objects (Objects wrapping the typical IEEE datatypes). 
We also have OPM (Object Persistence Mapping???) ... which you could use to annotate POJOs ... however this is a tool that hasen't received any updates for the last years. In theory it would make sense to combine the two ... to pass in an annotated class type and to receive populated objects, but that would be something someone would have to work on ... maybe you? ;-)

So I will be working on the EIP driver quite a lot in the next 1,5 months ... However can't I do any pormisses as another paid gig has priority over this.

However, if you want to help working on this, any help is greatly appreciated and I would be happy to help you.

Interested?

chrisdutz, 2023-05-23:
Ok ... making good progress on my ControlLogix setup ... however it seems that the Logix Designer in the Lite edition doesn't seem to support UDTs ... so guess I won't be able to test this feature.

chrisdutz, 2023-05-23:
Scrap that last message of mine ... thankfully @hutcheb knows his AB stuff and he pointed out that UDTs are a quite essential part of the Studio 5000 software and I actually found it ... it just seems that they moved the location of the UDTs from the top-level of the tree ...

chrisdutz, 2024-07-26:
So ... a little update on this: So in general we are already able to read UDTs, the problem is that on the side of PLC4X we don't know how to decode the data passed back from the server. In order to do this we need to ask the PLC to read the tag definitions as well as the data-type-definitions and decode them. This is what the PLC4X driver is missing. Out of curiosity, as I have an AB PLC myself, I have built my own version of the Logix driver that is able to read and write UDTs (Among other features, such as Auto-Discovery and Browsing),  however as I posted in my recent blog post, I have stopped working on PLC4X in my free time and will only donate stuff I'm doing in my free time, if I get enough support from the people using it in form of donations, hardware or whatsoever (Here the article for your reference: https://github.com/chrisdutz/blog/blob/main/plc4x/throwing-the-towel.adoc)

chrisdutz, 2024-08-16:
Closing for lack of activity.

closed 2024-08-16 by chrisdutz