apache/plc4x #972: [Feature Request]: PLC4X Kafka Connector without Confluent Hub

pm60lixy, 2023-06-01:
### What would you like to happen?

In my docker environment i have pulled the latest version of the plc4x kafka connector via the docker confluent hub in the past. Since it isn't available anymore how do i get the latest version as a plugin for the docker confluent control center? 

best regards

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [X] OPC-UA
- [X] S7

chrisdutz, 2023-06-01:
Hi,

the reason we decided to take the connector out of the confluence store was simply that Confluent kept on nagging us to do releases faster than the project wanted to (Whenever their tooling showed any vulnerable third party dependencies, they kept on opening issues with the Security team). The thing is almost nobody working on PLC4X is actually really being paid for our work. We simply didn't want to be pushed by a company earning money but giving nothing back to the community.  

If more companies using our stuff would consider giving back to the communities, I think we would have continued supporting stuff like the confluent store, but this way we decided to remove it from the store.

You should be absolutely fine with building PLC4X locally and simply copying the jar to your docker container ... or simply downloading a convenience binary from the Apache maven repository.

If you and your company need things to be in the confluent ecosystem ... you might consider funding someone willing to do the extra work for more frequent releases.

Chris

pm60lixy, 2023-06-01:
Hey Chris, 

thanks for the quick response! i really appreciate the work of the team and will speak with our company if its possible to fund this project. 

best regards

chrisdutz, 2023-07-21:
So ... we do have the connector and it should be easy to use it ... can we close this issue? 
If you look at: 
https://repo1.maven.org/maven2/org/apache/plc4x/plc4j-apache-kafka/0.10.0/

The file: https://repo1.maven.org/maven2/org/apache/plc4x/plc4j-apache-kafka/0.10.0/plc4j-apache-kafka-0.10.0-standalone.zip should be what you need to deploy in kafka-connect in order to get it working.

pm60lixy, 2023-07-21:
Thanks a lot, yes we can close it.

Best regards
Philipp

closed 2023-07-21 by chrisdutz