apache/plc4x #999: [Bug]:  Appears when reading or writing PLC values ： java.util.NoSuchElementException

cptjackwu, 2023-06-19:
### What happened?

When I execute a read or write operation at a fixed time of 300 milliseconds, the following error message will appear：

PlcClient.class  ，Wrong place to report
[image]

java.util.NoSuchElementException: null
	at java.util.AbstractQueue.remove(AbstractQueue.java:117)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.processWorklog(RequestTransactionManager.java:104)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.submit(RequestTransactionManager.java:99)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager$RequestTransaction.submit(RequestTransactionManager.java:178)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.write(S7ProtocolLogic.java:310)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.lambda$optimizedWrite$2(BaseOptimizer.java:113)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:150)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedWrite(BaseOptimizer.java:113)
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.write(AbstractPlcConnection.java:175)
	at org.apache.plc4x.java.spi.messages.DefaultPlcWriteRequest.execute(DefaultPlcWriteRequest.java:66)
	at com.wh.subscription.PlcClient.execute(PlcClient.java:275)
	at com.wh.subscription.PlcClient.writeBOOL(PlcClient.java:162)
	at com.wh.geelyuc.listener.HeartInputListener.lambda$null$0(HeartInputListener.java:56)
	at com.wh.common.function.FunctionalWrapper.lambda$runnableNullThrowWrapper$7(FunctionalWrapper.java:151)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset$$$capture(FutureTask.java:308)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

glcj, 2023-06-19:
Hello,

Can you try version 0.11.0-SNAPSHOT?

If the problem continues, can you tell me the PLC model, and a list of PlcTags that you require?

Kind regards,

sruehl, 2023-06-19:
Think that might be a bug @splatch fixed with cd5df7e5b43c6725d4e427181c85bb4b4d68699a

splatch, 2023-06-20:
Yup, this one should be gone in 0.11. We had a wrongly constructed loop. @cptjackwu this change can be backported to 0.10 if needed, cause this part of code did not receive many updates.

cptjackwu, 2023-07-01:
When will the stable version be released? about 0.11

chrisdutz, 2023-07-01:
We need to do quite a bit of cleaning up and refactoring for the most recent changes to s7 and eip ... I would say something within the next few months.

chrisdutz, 2023-07-21:
So can we consider this bug fixed in 0.11.0 and close this issue?

splatch, 2023-07-21:
Yes, it is part of develop and method call which generates this error is not present any more.

closed 2023-07-21 by splatch