apache/plc4x #1012: [Bug]: s7-1500 NullPointerException

anganing, 2023-06-30:
### What happened?

help me,please!
i want to read s7-1500, but s null pointer exception occurred! hear is my code:
[image]

the exception:
[image]

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

hongjinlin, 2023-06-30:
Hello, is your s7-1500 a real device?

glcj, 2023-06-30:
Hello,

1. If it is a physical equipment, check the address of the device or the CP.
2. If it is the Advanced PLCSIM, you must assign it an IP address within your subnet.
3. If it's not the advanced, standard PLCSIM, you can try [1].

1. https://nettoplcsim.sourceforge.net/

anganing, 2023-07-03:
@hongjinlin 
no, I use snap7 server demo.
[image]

anganing, 2023-07-03:
@glcj 
i use snap7 server demo. i repalce it to NetToPLCSim, when i connect to the server:
[image]

spnettec, 2023-07-03:
try my repository. It's NettyPlcConnection bug

anganing, 2023-07-03:
@spnettec how to use your repository?

anganing, 2023-07-03:
@spnettec the payload is null,It's NettyPlcConnection bug?
[image]

hongjinlin, 2023-07-03:
> @hongjinlin no, I use snap7 server demo. [image]

It seems that the snap7 server is not started successfully. Can you confirm if your 127.0.0.1:102 is listening?

anganing, 2023-07-03:
> > @hongjinlin no, I use snap7 server demo. [image]
> 
> It seems that the snap7 server is not started successfully. Can you confirm if your 127.0.0.1:102 is listening?

yes, i sure. that is i use another tool with port 102 cause Permission denied

hongjinlin, 2023-07-03:
> > > @hongjinlin no, I use snap7 server demo. [image]
> > 
> > 
> > It seems that the snap7 server is not started successfully. Can you confirm if your 127.0.0.1:102 is listening?
> 
> yes, i sure. that is i use another tool with port 102 cause Permission denied

Hi, I think you can use the snap7 client to determine whether the data can be read normally. Or if you just want to confirm whether the PLC4X is normally used in your scenario, you can try it with a physical device.

anganing, 2023-07-03:
> 

thanks a lot, I found the bug: invalid resource address. I think the plc4x's docs are not enough detailed,some times bugs given
[image]
[image]

closed 2023-07-03 by anganing

sruehl, 2023-07-03:
whatever happens a NPE shouldn't occur there. @glcj any idea?

glcj, 2023-07-04:
Hello,

I'll take note, but it must be associated with regex processing.

It can be seen that it has erroneous addresses, for security it should pre-process the PlcTags before carrying out the build and discard those that contain errors.

From that point of view it is not so unreasonable to avoid execution with error tags.

Kind regards.

anganing, 2023-07-05:
> Hello,
>
> I'll take note, but it must be associated with regex processing.
>
> It can be seen that it has erroneous addresses, for security it should pre-process the PlcTags before carrying out the build [...]

Thank you very much for your help. I hope the documentation can be more detailed (with more usage examples), which can help beginners quickly use it