apache/plc4x #1018: [Bug]: write a value to coil by modbus-tcp,respons:INVALID_DATA

anganing, 2023-07-06:
### What happened?

hello,
I use ModbusPal and Modbus Slave as a Slave.

when I write a coil value（true or 1） to slave,  response code is all INVALID_DATA.

addtionally, read coil all no problem.

hear is my code:

[image]
[image]
[image]

any advice?

Respectfully.

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2023-07-10:
I was able to replicate this using version 0.10.

It is fixed in the latest version 0.11 snapshot.

spnettec, 2023-07-10:
Did you set ModbusPal to Holding registers? If i set to Holding registers i have this error. If set Coils correctly it is ok.
[image]

anganing, 2023-07-10:
> I was able to replicate this using version 0.10.
> 
> It is fixed in the latest version 0.11 snapshot.

oh,thanks

closed 2023-07-10 by anganing