apache/plc4x #1031: [Feature Request]: How to handle reconnection after disconnection in a subscription?

anganing, 2023-07-13:
### What would you like to happen?

hello,

How to handle reconnection after disconnection in a subscription? I couldn’t find any relevant examples on your official website.
```
connection.addDisconnectListener(event -> {
    // Handling logic when the connection is lost
    System.out.println("Connection lost. Reconnecting...");

    // Perform reconnection operations here, such as reconnecting or restarting the subscription process
});
```

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2023-07-13:
Hello @anganing, I was facing similar issue with AMSADS, which lead to concept of (dis)connection listener.  I am not sure if this API is fully supported for 0.11 build thus I'd need first to test that.
With regard to subscriptions I believe we will face situation where diver and its internal state have them (and their handles/identifiers), however OPC UA server will not send any updates for them since connection for which these were made is gone.

chrisdutz, 2023-07-13:
Yeah... That's a thing we need in the driver core of subscription drivers, but not implemented yet. Currently lukasz' approach is probably the best.