apache/plc4x #1032: [Feature Request]: Can you simplify the driver development process?

anganing, 2023-07-17:
### What would you like to happen?

hello,
I am A Java developer and I only need to develop some drivers for plc4j such as focas, Qna-3E, A-1e, etc. When I built plc4x from source, I found that it was too bloated, can you simplify it or any advice?

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-07-17:
Well actually, if you're only building for java, the build is pretty simple.

But I am planning on creating some tutorial videos in the next months. So perhaps they will help you. Till then: feel free to ask questions on our mailinglist or using GitHub Discussions.

anganing, 2023-07-17:
Oh, that's great. Where will you post the video tutorials, I would like to subscribe in advance if possible.

Also suggest a video: How to develop your own driver, such as MyModbusTcp

Thank you very much

chrisdutz, 2023-07-17:
Stuff like that is on the roadmap ... however all parts have not been decided yet ... same with a concrete time-frame as well as where to host them. I would assume that it will be one of the usual locations.

hutcheb, 2023-07-18:
There's a lot of stuff in the repository, not all of it is absolutely needed in each case.  if you only need certain drivers then you just build those (and it's dependencies). Something like this will show you what artifacts are needed for a driver(s) and just compile the bare minimum.

[image]

I find when developing new drivers, I use the existing drivers as a base and basically copy the protocol and driver packages for a specific driver, and then just remove anything that's not needed for the new driver.

There's some info on the developer section of the website, but otherwise continue to ask questions and there's normally people that will help.

chrisdutz, 2024-08-16:
I think we can close this issue, right?

closed 2024-08-22 by anganing