apache/plc4x #1033: [Feature Request]: when 0.11.0 version release???

kdxq, 2023-07-19:
### What would you like to happen?

when 0.11.0 version release???

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-07-19:
when done ;-)

sruehl, 2023-07-19:
I really dislike those "-RC/Milestone/whatever" releases as those trigger renovate/dependabot and suggest those are normal releases as those bots/implementation don't understand that this isn't a real release. And by just releasing something it doesn't magically stabilize. E.g. we still need to fix the s7 driver in golang.

sruehl, 2023-07-19:
Also: We always have the option to backport changes into a point release if they a urgently needed.

sruehl, 2023-07-19:
so to wrap that up: we don't have a date for a release yet