apache/plc4x #1053: [Bug]: Time_waiting  many  threads causing server stuck

644036249, 2023-08-01:
### What happened?

[image]
[image]

Did I find that there are many times during the PLC4J testing_ The threads in the waiting state were generated when I was not connected to the PLC. I wanted to test and read data from the PLC, but these times_ Waiting will occupy memory and ultimately cause the laptop to crash. However, the connection was made using getConnect and was also abnormally closed using jdk8's try(). May I ask if this problem can be solved at present and how to solve it?

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

splatch, 2023-08-07:
Sorry but your report does not contain code sample, just pictures of it.
By checking pictures themselves it is clear to me that you open connection multiple times to the same device but never close it. Please implement proper resource management in your application by calling `connection.close()`.

Feel free to re-open issue in case if issue remains after above fixes are made.

closed 2023-08-07 by splatch

glcj, 2023-08-07:
Hello everyone,

As @splatch points out, it is recommended that you post sample code (test) in order to identify the problem.

You have to differentiate the problem between the Cache handler and the driver itself. The number of tasks that the driver raises is limited (of the Netty and those of the S7 driver itself).

In the version of the S7H driver found in [1], it identifies the thread pools in order to differentiate between them.

When you open the request again, add a screenshot of the jconsole to be able to identify your problem, which is already quite strange.

Kind regards,

1. https://github.com/apache/plc4x/tree/feature/hop