apache/plc4x #1054: [Feature Request]: I encountered some configuration issues when using PLC4X for the first time

kuikuijiademao, 2023-08-01:
### What would you like to happen?

```
 PlcReadRequest.Builder intReadBuilder = plcConnection.readRequestBuilder();
            intReadBuilder.addItem("intData", "holding-register:510:INT[1]");
            PlcReadRequest intReadRequest = intReadBuilder.build();
            PlcReadResponse intReadResponse = intReadRequest.execute().get();
```
I want to read data at position 510 in register 03. The value in PLC is 1, but I have read a value of 0. I would like to know how to set the parameter information in the code through the modbus TCP protocol

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-08-01:
Try reading 509 or 511, if that contains the expected value ... Modbus is strange when it comes to these offsets (Not quite sure in which direction the offset was, that's why I'm asking you to double check)
The register 1 actually has the address 0 on the wire ... that's why I'm asking.

chrisdutz, 2023-09-22:
Is this still an issue?

chrisdutz, 2023-09-29:
But I'm also seeing that you are using the latest released version, as we refactored the API quite a bit. I have tested your usecase and it's returning the right information. We are going to release PLC4X 0.10.0 in a few days, so please re-try with that.

closed 2023-09-29 by chrisdutz