apache/plc4x #1068: [Bug]: Once a connection is established using modbus tcp, it cannot be reconnected once disconnected

StrawberryBlue, 2023-08-09:
### What happened?

Once a connection is established using modbus tcp, it cannot be reconnected once disconnected.Or where is the reconnect demo, am I using it the wrong way?

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-09-22:
Sorry for the late response ... in order to tell you if you're using it wrong, I guess you'd have to show us how you're using it.
In general it's strange, as Modbus is stateless and the concept of a "connection" is unknown to the protocol. Perhaps it's a problem with the TCP socket re-usage on your device (My KNX for example has a timeout of 5 minutes after disconnecting)

chrisdutz, 2023-09-29:
So ... for the sake of being right here, I wrote this little program, and it worked as it should ... So I guess you were doing it wrong. But at least you can use this example for knowing how it was supposed to be used (However I wouldn't reccomend this pattern in general ... wir modbus it's ok, but with other drivers such as ADS, this approach would be very imperformant).

```
    public static void main(String[] args) throws Exception {
        try (PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection("modbus-tcp://127.0.0.1")) {
            PlcReadRequest readRequest = plcConnection.readRequestBuilder().addTagAddress("Bool", "holding-register:1000:BOOL").build();
            PlcReadResponse readResponse = readRequest.execute().get();
        }
        try (PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection("modbus-tcp://127.0.0.1")) {
            PlcReadRequest readRequest = plcConnection.readRequestBuilder().addTagAddress("Bool", "holding-register:1000:BOOL").build();
            PlcReadResponse readResponse = readRequest.execute().get();
        }
    }

```

closed 2023-09-29 by chrisdutz