apache/plc4x #1089: [Bug]: Module not imported after following the guide

p3tr0v, 2023-08-30:
### What happened?

Even following the Starting guide, the module is no imported to project.

**Steps to reproduce in your go workspace:**
$ mkdir plcproject
$ go mod init
$ go mod tidy
$ go work use . 
_create main.go file with your IDE_
$ go get github.com/apache/plc4x/plc4go
_I can see ~/go/pkg/mod/github.com/apache/plc4/plc4go@v0.10.0_
Then after following the guide with 
`driverManager := plc4go.NewPlcDriverManager()`
It just cant find/import the module

**go.mod**
```
module plcproject

go 1.20

require github.com/apache/plc4x/plc4go v0.0.0-20230829070645-b2f3acb97b3b // indirect
```
In the project I can import any module from my _go/pkg/mod/github.com_ directory, but plc4go cant be imported, why?

### Version

v0.10.0

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

sruehl, 2023-08-30:
what does your import statement look like in the `main.go`? Can you post the code of that file?

sruehl, 2023-08-30:
delete the `go.work` and use this main.go:
```golang
package main

import plc4go "github.com/apache/plc4x/plc4go/pkg/api"

func main() {
  driverManager := plc4go.NewPlcDriverManager()
  println(driverManager)
}
```
It should output this when run with that `go run main.go`:
```
$ go run main.go
{"level":"trace","time":"2023-08-30T07:51:07+02:00","message":"Creating plc driver manager"}

```

p3tr0v, 2023-08-30:
> what does your import statement look like in the `main.go`? Can you post the code of that file?

thanks for your time!
The main is simple, when trying to code, VScode just dont find the module.
[image]
If I try to force the import, I got this
[image]

 My `go env`
```
GO111MODULE="on"
GOARCH="amd64"
GOBIN="/opt/golang/current/bin"
GOCACHE="/opt/golang/current/gocache/go-build"
GOENV="/home/user/.config/go/env"
GOEXE=""
GOEXPERIMENT=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOINSECURE=""
GOMODCACHE="/home/user/dev/projects/golang-workspace/pkg/mod"
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/home/user/dev/projects/golang-workspace"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/opt/golang/current"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/opt/golang/current/pkg/tool/linux_amd64"
GOVCS=""
GOVERSION="go1.20.5"
GCCGO="gccgo"
GOAMD64="v1"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD="/dev/null"
GOWORK=""
CGO_CFLAGS="-O2 -g"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-O2 -g"
CGO_FFLAGS="-O2 -g"
CGO_LDFLAGS="-O2 -g"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -Wl,--no-gc-sections -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build2976824462=/tmp/go-build -gno-record-gcc-switches"
```

whatever other project I can import from ~/go/pkg/mod/github.com, so the problem isn't the IDE.

Why should I delete go.work? Even without this file, the issue persists.

> and what starting guide are you referring to?

https://plc4x.apache.org/users/getting-started/plc4go.html

p3tr0v, 2023-08-30:
reopening

sruehl, 2023-08-30:
I could not get it working with the `go.work` at I don't know what it does. So if I follow your steps without the `go work use .` (which doesn't work if you strictly follow your steps).
Do the following:

1. go to tmp `cd /tmp`
2. make your folder `mkdir plcproject`
3. go into your folder `cd plcproject`
4. init the module `go mod init plcproject`
5. do a tidy `go mod tidy`
6. get plc4go `go get github.com/apache/plc4x/plc4go`
7. paste the `main.go` I posted above https://github.com/apache/plc4x/issues/1089#issuecomment-1698535753: `vi main.go`
8. do a tidy `go mod tidy`
9. run the main `go run main.go`

sruehl, 2023-08-30:
(also I might update the documentation if I find some time or create a PR if you want integrating the steps I posted above)

sruehl, 2023-08-30:
yes, but as you can see it works for you. The pcap error is because you need pcap on your system

p3tr0v, 2023-08-30:
> apt-get install libpcap

Got success after installing.

But I have to understand why without go.work, from my golang workspace projects, my IDE still saying it:

[image]

Doesnt make sense to me.....

sruehl, 2023-08-30:
no sorry can't help you with that :(

p3tr0v, 2023-08-30:
thank you for your time.

closed 2023-08-30 by p3tr0v