apache/plc4x #1101: [Bug]: plc4j-driver-opcua seems to leak memory

takraj, 2023-09-23:
### What happened?

The OPC-UA driver seem to leak memory when closing a connection object. This issue sooner or later lead to application crash with OOM error. Workaround would be to reuse connection objects, but then we face the issue #1100, which I believe renders a fix for this pretty urgent.

[image]

Example code to reproduce the issue:
```
PlcDriverManager driverManager = new PlcDriverManager();

AtomicBoolean running = new AtomicBoolean(true);
Runtime.getRuntime().addShutdownHook(new Thread(() -> running.set(true)));

while (running.get()) {
    PlcConnection connection = driverManager.getConnection("opcua:tcp://127.0.0.1:40185/milo?discovery=false");
    try {
        Thread.sleep(100);
    } finally {
        connection.close();
    }
}
```

Some additional charts
--------------------------------

[image]

[image]

[image]

[image]

Logs
--------

[logs-before-oom.txt](https://github.com/apache/plc4x/files/12706498/logs-before-oom.txt)

Heap dumps
-------------------

* [MemoryLeakDemo_1045304_23_09_2023_12_41_27.zip](https://github.com/apache/plc4x/files/12706514/MemoryLeakDemo_1045304_23_09_2023_12_41_27.zip)
* [MemoryLeakDemo_1045304_23_09_2023_13_02_21.zip](https://github.com/apache/plc4x/files/12706515/MemoryLeakDemo_1045304_23_09_2023_13_02_21.zip)
* [MemoryLeakDemo_1045304_23_09_2023_14_05_15.zip](https://github.com/apache/plc4x/files/12706519/MemoryLeakDemo_1045304_23_09_2023_14_05_15.zip)

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2023-09-29:
I'll have a look at this as it seems none else is currently planning on doing so ... hopefully this doesn't require too much knowledge of the OPC-UA protocol.

chrisdutz, 2023-09-29:
Having seen now that you were using the pre 0.10.0 syntax for creating the connection, I think you have stumbled over a problem that we have already addresses in 0.10.0-SNAPSHOT ... we're planning on releasing that in the next few days. Would be cool, if you could have a look if this is fixed for your case too. 

I am currently running an updated version of your program in JProfiler and am not seeing any increase in memory usage.

chrisdutz, 2023-09-29:
[image]

chrisdutz, 2023-09-29:
Ok ... after running it for almost two hours ... there's definitely something going on ... not sure if it's because I'm also running the server in the same VM ... It's consuming quite a bit more Memory and a lot more CPU time.
[image]

chrisdutz, 2023-09-29:
Ok ... just noticed I was running it in the IntelliJ profiler, not JProfiler ... I split up the application into a server (that just starts milo) and a client (with your code) ... when profilling only the client I can see a constant increase in sleeping threads ... I should probably figgure out why this is happening and fix it before the 0.11.0 release ...
[image]

chrisdutz, 2023-09-29:
And they all seem to be related to the nio event group: 

[image]

Hmpf ... I thought we had addressed that issue and I guess it probably also has an effect on all other drivers too ... Usually opening and closing connections shouldn't be the default case, as the connection-cache should be used, but I think having somethign stuck will also be the reason why some applications simply don't stop gracefully.

chrisdutz, 2023-09-29:
[image]
 Should have been this image

chrisdutz, 2023-09-29:
YAY ... I found it :-)
It wasn#t the event loops .. they are correctly closed.
It was the NettyHashTimerTimeoutManager that we start for every connection but never explicitly close it ... by overriding the close() method of ChannelDuplexHandler in Plc4xNettyWrapper and explicitly closing it, the leak of open threads is gone ... you can see a commons-pool growing at the start, but as soon as it's reached it's 11 threads the thread-count stays constant :-)

chrisdutz, 2023-09-29:
[image]

chrisdutz, 2023-09-29:
Please give the current 0.11.0-SNAPSHOT version a try and check if the problem is gone.

takraj, 2023-10-10:
@chrisdutz I have tried the revision tagged with [v0.11.0](https://github.com/apache/plc4x/releases/tag/v0.11.0), but the problem does not seem to be resolved yet. :(

```
~/plc4x$ git log -1
commit d22042ef079ae93120770a95704571118d9c871e (HEAD, tag: v0.11.0)
Author: Christofer Dutz <cdutz@apache.org>
Date:   Mon Oct 2 09:51:59 2023 +0200

    [maven-release-plugin] prepare release v0.11.0
```

As you can see on the screenshot below, the heap usage is still constantly increasing:
[image]

Tested with the following code:
```java
public class MemoryLeakDemo {

    public static void main(String[] args) throws Exception {
        PlcDriverManager driverManager = new DefaultPlcDriverManager();
        PlcConnectionManager connectionManager = driverManager.getConnectionManager();

        AtomicBoolean running = new AtomicBoolean(true);
        Runtime.getRuntime().addShutdownHook(new Thread(() -> running.set(true)));

        while (running.get()) {
            PlcConnection connection = connectionManager.getConnection("opcua:tcp://127.0.0.1:12686/milo?discovery=false");
            try {
                Thread.sleep(100);
            } finally {
                connection.close();
            }
        }
    }
}
```

qtvbwfn, 2023-12-09:
> ### What happened?
> The OPC-UA driver seem to leak memory when closing a connection object. This issue sooner or later lead to application crash with OOM error. Workaround would be to reuse connection objects, but then we face the issue #1100, which I believe renders a fix for this pretty urgent.
> 
> ![Screenshot from 2023-09-23 14-14-10](https://private-user-images.githubusercontent.com/2440418/270109529-17d5329b-e1ed-459c-bd1f-5d3339b22982.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMDg2ODcsIm5iZiI6MTcwMjEwODM4NywicGF0aCI6Ii8yNDQwNDE4LzI3MDEwOTUyOS0xN2Q1MzI5Yi1lMWVkLTQ1OWMtYmQxZi01ZDMzMzliMjI5ODIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDlUMDc1MzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWJhNzhlZjEzM2NlMzMxZjQyZTc2MWY5YzQxNGNjMzU4YzliMjIyNjdiMWZjMTllMmQ1MTZiMzM1MDRhZDRlYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Or8mol9RudsyWBXoXcLdNGRsuCZqq7ephJ90EK2hadE)
> 
> Example code to reproduce the issue:
> 
> ```
> PlcDriverManager driverManager = new PlcDriverManager();
> 
> AtomicBoolean running = new AtomicBoolean(true);
> Runtime.getRuntime().addShutdownHook(new Thread(() -> running.set(true)));
> 
> while (running.get()) {
>     PlcConnection connection = driverManager.getConnection("opcua:tcp://127.0.0.1:40185/milo?discovery=false");
>     try {
>         Thread.sleep(100);
>     } finally {
>         connection.close();
>     }
> }
> ```
> 
> ## Some additional charts
> ![Screenshot from 2023-09-23 14-23-14](https://private-user-images.githubusercontent.com/2440418/270109655-1384d3d8-63b8-4abb-ac75-4843d9bf72b2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMDg2ODcsIm5iZiI6MTcwMjEwODM4NywicGF0aCI6Ii8yNDQwNDE4LzI3MDEwOTY1NS0xMzg0ZDNkOC02M2I4LTRhYmItYWM3NS00ODQzZDliZjcyYjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDlUMDc1MzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjJkOTMwOTllZTgzYmZhNTA5Mzc2ZDM4MzkwODVkNjk1YjczZjNhNDJlN2QyNDY1ZDhkOTlkNTdlNTgzMjMxOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pnRLv3al4dwtHBIkkc1d7GZhPNJfg0VCbzhDZybdlAI)
> 
> ![Screenshot from 2023-09-23 14-08-05](https://private-user-images.githubusercontent.com/2440418/270109667-0a50afb6-3d8d-436f-afcb-a3bb7253f1f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMDg2ODcsIm5iZiI6MTcwMjEwODM4NywicGF0aCI6Ii8yNDQwNDE4LzI3MDEwOTY2Ny0wYTUwYWZiNi0zZDhkLTQzNmYtYWZjYi1hM2JiNzI1M2YxZjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDlUMDc1MzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTg4YmU3MmEyNzBhOTBiYTViOWEwNDE3ZGRhNDI0OWEzOTJjNzdjNjIyZTZmZTc5MGQ4OTcyZmY4ZmU2OGY5MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.aPs8eXTiltN8VRlCpJFGKcmgiufK4R3EnHcbgw4iJyA)
> 
> ![Screenshot from 2023-09-23 14-10-16](https://private-user-images.githubusercontent.com/2440418/270109932-e98e765c-79a5-40bc-b342-4c4acd9c9337.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMDg2ODcsIm5iZiI6MTcwMjEwODM4NywicGF0aCI6Ii8yNDQwNDE4LzI3MDEwOTkzMi1lOThlNzY1Yy03OWE1LTQwYmMtYjM0Mi00YzRhY2Q5YzkzMzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDlUMDc1MzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjdjNzQ3NWQzYWMyODBkM2M2YzM4N2M1ODhhYmZmODc2OTJlZmM5MjBiZGZkOTc2MTA0M2RlZGRkODQzZGNjYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.UX_JnzyOxxLhhrS3Sarbn0MYR_Y5KyzIVGXD7Ce2qFI)
> 
> ![Screenshot from 2023-09-23 14-10-25](https://private-user-images.githubusercontent.com/2440418/270109692-913b2138-0571-4cff-a75b-9ebedebb1f56.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMDg2ODcsIm5iZiI6MTcwMjEwODM4NywicGF0aCI6Ii8yNDQwNDE4LzI3MDEwOTY5Mi05MTNiMjEzOC0wNTcxLTRjZmYtYTc1Yi05ZWJlZGViYjFmNTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDlUMDc1MzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDcxYjVjMTBiN2YyYzk3MTIyZmExNWY5ZTZhMjA4N2MxZTk2ZmIxOWI1YzZkYjk3N2I5YjZlNDc1MmY1YTE0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.6MuyzqlNR9Fj74qe2hbD5Y2GWbQrgAPZrr_k5bptWfw)
> 
> ## Logs
> [logs-before-oom.txt](https://github.com/apache/plc4x/files/12706498/logs-before-oom.txt)
> 
> ## Heap dumps
> * [MemoryLeakDemo_1045304_23_09_2023_12_41_27.zip](https://github.com/apache/plc4x/files/12706514/MemoryLeakDemo_1045304_23_09_2023_12_41_27.zip)
[... cut ...]
> * [ ]  BACnet/IP
> * [ ]  CANopen
> * [ ]  DeltaV
> * [ ]  DF1
> * [ ]  EtherNet/IP
> * [ ]  Firmata
> * [ ]  KNXnet/IP
> * [ ]  Modbus
> * [x]  OPC-UA
> * [ ]  S7
hi:
    It took me about three days, and I have fixed the thread and memory overflow issues in the V0.11.0 version. Can you create a new branch 0.11.1-SNAPSHOT for me to submit my code?
 The main issue is that the keepAlive in SecureChannel causes the channel to be occupied for a long time without being released.
@chrisdutz @takraj @hutcheb 
[image]

chrisdutz, 2023-12-09:
I the world of plcs generally connections should be reused, as for most connections establishing a connection is pretty cost intensive. I know that opc-ua needs to resolve some things when connecting. 

So I agree, that both issues should be addressed.

Just not sure when. If nobody else takes on the issue, generally opc-ua related issues are pretty low on my personal priority list 😉

qtvbwfn, 2023-12-09:
[image]
[image]
[SecureChannel.java.txt](https://github.com/apache/plc4x/files/13623416/SecureChannel.java.txt)

christofe-lintermans-actemium, 2024-01-02:
Hi All,

I have the same issue, but when using Apache NiFi integration.

[image]

Should I create a new issue, or is this issue 1101 sufficient?

Could a priority be assigned to address this issue?

Thank you in advance!

chrisdutz, 2024-01-02:
I would strongly assume it's the same issue, as from my knowledge interenally all integration modules use the same components. So this issue should probably be observalble in all integration modules.

chrisdutz, 2024-01-02:
Aaaah ... cool ... well in that case ... I guess as soon as these changes make it back to develop, sounds like a good time for releasing ;-) (I guess we've got enough stuff in there already)

christofe-lintermans-actemium, 2024-04-11:
Hi everyone,

I have updated PLC4X to version 0.12 which contains the commits for #1007. Unfortunately, the memory leak issue persists.

[image]

Does anyone have insights into the root cause of this problem? I am willing to investigate further.

I aim to use PLC4X exclusively as the source/sink library for all my data projects. For S7 and Modbus protocols, the performance and stability are satisfactory, if not for occasional connection interruptions but are handled by NiFi's timeout settings. However, the Out of Memory (OOM) issue with OPC UA is particularly problematic.

Kind regards,

Christofe

qtvbwfn, 2024-04-12:
I previously attempted to address this issue on version 0.10.0, where the main problem resided in OpcuaProtocolLogic. It occurred when calling the close method on the PlcConnection, resulting in resources not being properly released. Subsequently, due to significant API changes, I did not pursue further investigation at that time. Since then, for OPCUA handling, I have consistently used the Eclipse Milo client.

@splatch

christofe-lintermans-actemium, 2024-04-12:
> I previously attempted to address this issue on version 0.10.0, where the main problem resided in OpcuaProtocolLogic. It occurred when calling the close method on the PlcConnection, resulting in [...]

Thank you for the info.
PLC4X uses eclipse milo underlying for the OPCUA driver.

[image]

So I want to avoid a proprietary implementation for OPC UA.

splatch, 2024-04-12:
> PLC4X uses eclipse milo underlying for the OPCUA driver.

@christofe-lintermans-actemium It used to in 0.8 release. Since 0.10 there is in-house implementation of PLC4X driver which rely on same infrastructure as S7 and Modbus drivers. 

@qtvbwfn thank you for the hint, I'll check what's going on there!

splatch, 2024-04-12:
I see where the gap between s7 and opcua is, working on test right now.

qtvbwfn, 2024-04-12:
[image]
[image]
By testing the latest version and tracking the logs of the OPC UA SERVER, it was found that two connections were created each time, but only one was closed upon termination.

@splatch @chrisdutz @takraj @hutcheb

splatch, 2024-04-12:
Looks like pre-flight discovery connection might be part of the trouble.

splatch, 2024-04-12:
Please have a look on PR I just created, it makes sure that discovery connection, transaction manager and a lot of threads gets shut.

Chris fixes from few months ago worked fine, however while working on UA security I fixed discovery connection which made its resources hang again. With above fix thread count stays flat, even if new connection is created every second:

[image]

christofe-lintermans-actemium, 2024-04-12:
@splatch 

Can you guide me on how to build the nar file for NiFi including this fix?  I already checked out the branch locally.
I can quickly test it on the system.

Thank you in advance.

Christofe

chrisdutz, 2024-04-12:
Yes ... we split up the repo to separate the integration modules, additional tools and examples from the main repository.

closed 2024-04-12 by splatch

christofe-lintermans-actemium, 2024-04-19:
Hi all,

I have checked out the specific branch:

[image]

I have built the nifi integrations and deployed it:

[image]

I didn't had any OOM problems anymore

[image]

(red line is time when I replaced the plc4x nar)

Thank you very much for your support, much appreciated!

commit f30412c0 (2024-04-12): Cleanup discovery connection resources.

Update of unit test to verify method exits.

Closes #1101.
PR #1518 (2024-04-12): Cleanup discovery connection resources.
Closes #1101.
commit 7f40bbba (2024-04-12): fix(plc4j/opcua): Cleanup discovery connection resources. 

Update of unit test to verify method exits.

Closes #1101.