apache/plc4x #1102: [Bug]: plc4j-driver-opcua - No subscription events are generated for `addCyclicField(...)` tags

takraj, 2023-09-23:
### What happened?

The documentation says that tags added via `addCyclicField(...)` will be polled in the specified time intervals. My expectation is to get subscription events each time these fields are getting polled, but apparently I don't get any. The other two types of subscriptions are working OK, however I don't see any difference between `addChangeOfStateField(...)` and `addEventField(...)` with the OPC-UA driver.

Example code to reproduce the issue:
```
PlcDriverManager driverManager = new PlcDriverManager();
try (PlcConnection opcuaConnection = driverManager.getConnection("opcua:tcp://opcuaserver.com:48010")) {
    PlcSubscriptionRequest request = opcuaConnection.subscriptionRequestBuilder()
            .addCyclicField(
                    "Demo",
                    "ns=2;s=Demo.Static.Scalar.String",
                    Duration.ofSeconds(1)
            )
            .build();

    PlcSubscriptionResponse response = request.execute().get();
    if (response.getResponseCode("Demo") != PlcResponseCode.OK) {
        throw new RuntimeException("Not OK.");
    }

    AtomicLong counter = new AtomicLong();

    response.getSubscriptionHandle("Demo").register(
            subscriptionEvent -> {
                counter.incrementAndGet();
                System.out.println(subscriptionEvent);
            }
    );

    Thread.sleep(10 * 1000); // 10 seconds

    if (counter.get() == 0) {
        throw new RuntimeException("No events received.");
    }
}

System.out.println("OK");
```

Logs
-------

* [logs.txt](https://github.com/apache/plc4x/files/12706657/logs.txt)

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2023-09-29:
@hutcheb or @sruehl ... I know you two are involved a bit more in this protocol ... can you help here?

hutcheb, 2023-10-01:
So it turns out the OPCUA subscriptions  don’t support a poling interval (Return a value every X seconds). They only support returning values when they change.

We would have to create an internal register of values, update it every time we see values change, and then use the internal value to generate a subscription response every X seconds.

chrisdutz, 2023-10-01:
I guess building a general purpose subscription simulator that can be used for any driver would be a good idea.