apache/plc4x #1103: [Feature Request]: Implement `PlcConnection.ping()`

takraj, 2023-09-23:
### What would you like to happen?

Currently the `ping()` method only seems to be implemented in `plc4j-driver-s7`. In all other drivers it throws an exception, when called.

Please implement it for all drivers, where this is technically possible, so the user application could rely on this standard interface to test if a connection is broken, or it could use it for keeping a connection alive without actually doing real traffic.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [X] AB-Ethernet
- [X] ADS /AMS
- [X] BACnet/IP
- [X] CANopen
- [X] DeltaV
- [X] DF1
- [X] EtherNet/IP
- [X] Firmata
- [X] KNXnet/IP
- [X] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2023-09-25:
I guess for most drivers it should be quite simple ... however it probably should be a call that comes at as little as much costs ... do you happen to know for some of the protocols which requests could act as such low-cost requests?

takraj, 2023-10-10:
@chrisdutz Unfortunately I'm not familiar with these protocols. The sole reason of adding my contributions is that, I'm working at a company who has this library integrated in one of their solutions, and there were issues at a customer that I had to troubleshoot.

chrisdutz, 2024-02-06:
I have started adding ping implementations to some of the PLC4J drivers ... please have a look.

chrisdutz, 2024-02-10:
So far the following drivers should be supporting the PING operation:
- ADS
- KNXNet/IP
- Modbus(TCP/RTU/ASCII)
- S7
- Simulated

janb93, 2024-02-14:
Hi,
I have tried the ping with Modbus TCP using 0.12.0-SNAPSHOT, but I receive null as a response. See the code below:

```
      CompletableFuture<? extends PlcPingResponse> future = plcConnection.ping();
      future.whenComplete((response, throwable) -> {
          try {
              System.out.println("Response " + response);
          } catch (Exception e) {
              System.out.println("Throwable " + throwable);
            e.printStackTrace();
          }
      });
```

chrisdutz, 2024-02-14:
I was using a request, that the Modbus spec indicated as being mandatory, however have I seen that there are almost no Modbus devices that actually support it.

But the good thing is ... all of them respond with the same valid Error of "Illegal Function" or whatever ... so you can actually use it to ping if a device is responsive, however there is no always valid register, coil or whatsoever. So are you getting errors from the ping request, or are you getting OK status back?

janb93, 2024-02-15:
In "throwable" I receive java.util.concurrent.TimeoutException

chrisdutz, 2024-02-15:
Ok ... that's odd ... cause I thought the devices would simply return an error ... could you please send me a wireshark recording of this?

Perhaps we should switch to simply reading the holding-register:1 as I would assume that this should be implemented in any device.

janb93, 2024-02-15:
This is the Wireshark summary. The request is sent, but I receive no response

```
Modbus/TCP	65	   Query: Trans:     1; Unit: 255, Func:  43/  1: Read Device Identification

Modbus/TCP
    Transaction Identifier: 1
    Protocol Identifier: 0
    Length: 5
    Unit Identifier: 255
Modbus
    .010 1011 = Function Code: Encapsulated Interface Transport (43)
    MEI type: Read Device Identification (14)
    Read Device ID: Basic Device Identification (1)
    Object ID: VendorName (0)
```

chrisdutz, 2024-02-15:
Could you please attach the pgap or pcapng file here?
You sent the request ... I'm more interested in the response (Or was there no response at all? ... in that case, don't worry)

janb93, 2024-02-15:
There was no response, sorry

chrisdutz, 2024-02-15:
Well that's important information ... let me quickly change the ping operations to reading the holding-register:1

chrisdutz, 2024-02-15:
Ok ... I just pushed a change, that might help resolve this issue :-)

So I've added a "pingAddress" which defaults to 0x40001 (Holding Register 1). It will use this to do the ping operation. If the default causes problems with your modbus device, you can simply override it in the connection string by providing a "ping-address={someotheraddress}" option.

janb93, 2024-02-15:
Have you updated the SNAPSHOT version or how can I test this feature as simple as possible?

chrisdutz, 2024-02-15:
It's currently deploying the snapshots: https://ci-builds.apache.org/job/PLC4X/job/PLC4X/job/develop/1877/console

janb93, 2024-02-15:
I can confirm that now it works! Thank you!

chrisdutz, 2024-02-15:
YAY! ... so time to close this ... will probably be released in a few days ;-)

closed 2024-02-15 by chrisdutz