apache/plc4x #1106: [Feature Request]: plc4j-driver-opcua - Implement browsing

takraj, 2023-09-24:
### What would you like to happen?

The OPC-UA protocol supports browsing, but the driver does not. Please implement it, in order to be able to recursively fetch data with the library.

Example code to browse the value tree:
```
DefaultPlcDriverManager driverManager = new DefaultPlcDriverManager();
try (PlcConnection opcuaConnection = driverManager.getConnection("opcua:tcp://opcuaserver.com:48010")) {
    PlcBrowseRequest request = opcuaConnection.browseRequestBuilder()
            .addQuery("Folder","ns=2;s=Demo.Static")
            .build();

    PlcBrowseResponse response = request.execute().get();
    System.out.println(response.getValues("Folder"));
}
```

The above program throws the exception below:
```
org.apache.plc4x.java.api.exceptions.PlcUnsupportedOperationException: The connection does not support browsing
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.browseRequestBuilder(AbstractPlcConnection.java:159)
	at org.apache.plc4x.java.opcua.ManualOpcuaFolderNode.main(ManualOpcuaFolderNode.java:32)
```

Example output of another software, that supports browsing:
```
$ uabrowse -u "opc.tcp://opcuaserver.com:48010" -n "ns=2;s=Demo.Static"
WARNING:asyncua.client.client:Requested session timeout to be 3600000ms, got 600000ms instead
Browsing node ns=2;s=Demo.Static at opc.tcp://opcuaserver.com:48010

DisplayName                    NodeId                    BrowseName                Value                    

LocalizedText(Locale='', Text='Arrays') ns=2;s=Demo.Static.Arrays 2:Arrays                 
LocalizedText(Locale='', Text='Scalar') ns=2;s=Demo.Static.Scalar 2:Scalar                 
LocalizedText(Locale='', Text='VariableWithNumericNodeId') ns=2;i=4294967294         2:VariableWithNumericNodeId, 12 
LocalizedText(Locale='', Text='VariableWithGuidNodeId') ns=2;g=5ce9dbce-5d79-434c-9ac3-1cfba9a6e92c 2:VariableWithGuidNodeId , 0  
LocalizedText(Locale='', Text='VariableWithByteStringNodeId') ns=2;b=0x61626364         2:VariableWithByteStringNodeId, 0  
LocalizedText(Locale='', Text='Matrix') ns=2;s=Demo.Static.Matrix 2:Matrix
```

[Wireshark capture of the above browsing request](https://github.com/apache/plc4x/files/12708323/opc-ua-browse-request.pcapng.gz)

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2023-09-25:
Hi @takraj 

thanks for all the work on the OPC-UA driver you are currently working on ... highly appreciated. 
Could you do us a favor and sign up on our mailing list dev@plc4x.apache.org (by sending an empty email to dev-subscribe@plc4x.apache.org) and tell us a bit about you, what you're working on ... we'd love to get you in touch with the rest of our community.

Chris

takraj, 2023-10-10:
@chrisdutz I have subscribed to the mailing list, thanks for inviting me. Although I don't plan to do frequent contributions, because I don't really have the time for that. The reason I have been involved is, that I had to troubleshoot a product, that was built on this library at the company I'm working at. I don't think I'll add/fix things having large impact, only low hanging ones maybe.

chrisdutz, 2026-08-16:
Implemented on `develop`, so this is fixed.

`browseRequestBuilder()` on an OPC-UA connection now works instead of throwing
`PlcUnsupportedOperationException`. The query is the address of the node to start from, and an empty
query starts at the Objects folder:

```java
PlcBrowseResponse response = connection.browseRequestBuilder()
    .addQuery("hw", "ns=2;s=HelloWorld")
    .build().execute().get();

for (PlcBrowseItem item : response.getValues("hw")) {
    System.out.println(item.getName() + " -> " + item.getTag().getAddressString());
}
```

The browse is recursive, which is what you asked for: each item carries its children, the tree is
walked down from the starting node, and a visited set breaks cycles and shared nodes so a node is
listed but not expanded twice. Items come back with the server-side attributes resolved - data type,
readable/writable flags and array dimensions - which are fetched in a batch per level rather than
one request per node. `browseWithInterceptor` is supported as well if you want to filter or stop
early rather than pull the whole tree.

It is covered by the driver test suite against the example server, browsing both from the Objects
folder and from a specific node.

Closing as fixed.

closed 2026-08-16 by chrisdutz