apache/plc4x #1138: [Bug]: plc4j-driver-opcua - Keepalive threads are never shut down -> DEADLOCK

takraj, 2023-10-10:
### What happened?

This issue affects both `v0.10.0` and the pre-release `v0.11.0`, and also highlights a conceptual, architectural problem in the `plc4j` library in general, that applies to multiple drivers.

## Base problem

The OPC-UA driver starts a keepalive thread once a connection is successfully estabilished. One keepalive thread per connection, here: [SecureChannel.java:577](https://github.com/apache/plc4x/blob/v0.11.0/plc4j/drivers/opcua/src/main/java/org/apache/plc4x/java/opcua/context/SecureChannel.java#L577). The keepalive loop itself is created in the common thread pool (via `supplyAsync()`), which is limited in size, and shared between the library and the user application: [SecureChannel.java:998](https://github.com/apache/plc4x/blob/v0.11.0/plc4j/drivers/opcua/src/main/java/org/apache/plc4x/java/opcua/context/SecureChannel.java#L998). The exit condition of this loop is not defined, therefore it keeps running forever, and holds/blocks a thread in the common pool, even after the connection is closed. As the common pool is size limited (`nr_cpus - 1`), after a few iterations of connect/disconnect, the common pool gets exhausted, and it can no longer execute any new tasks. Neither from the plc4j library, nor from the user application. As `supplyAsync()` is also used for creating a subscription monitor loop, this means that loop never gets started, because of the keepalive threads: [OpcuaSubscriptionHandle.java:227](https://github.com/apache/plc4x/blob/v0.11.0/plc4j/drivers/opcua/src/main/java/org/apache/plc4x/java/opcua/protocol/OpcuaSubscriptionHandle.java#L227). This practically means that the future object returned by `PlcSubscriptionRequest.execute()` never completes.

And let's assume for a moment, that the base problem above is fixed, and the keepalive thread now has a proper exit condition. In this case, the use of common pool will limit the count of ongoing subscriptions in the application, as each subscription thread occupies a slot in the pool. If the user has 4 CPU cores, this means they cannot have more than 2 active subscriptions: 1 slot is held by keepalive, 2 slots are held by the subscription threads.

## Conceptual problem

Thread pools are not for long running tasks. They are for frequent, ephemeral calculations, to save the cost of spawning a new thread every time. I think the current (ab)use of common pool should be either redesigned, to avoid hosting infinite loops, or these should be replaced by dedicated threads.

## Example application to reproduce the Base problem

```java
public class ForkJoinPoolDemo {

    public static final String FJP_PARALLELISM = "java.util.concurrent.ForkJoinPool.common.parallelism";

    public static void main(String[] args) throws Exception {
        System.out.println(FJP_PARALLELISM + ": " + System.getProperty(FJP_PARALLELISM));
        System.out.println("CPU Core: " + Runtime.getRuntime().availableProcessors());

        printDetails();

        PlcDriverManager driverManager = new DefaultPlcDriverManager();
        PlcConnectionManager connectionManager = driverManager.getConnectionManager();

        for (int i = 0; i < 50; i++) {
            Thread.sleep(1000);

            try (PlcConnection connection = connectionManager.getConnection("opcua:tcp://127.0.0.1:12686/milo")) {
                printStats(format("Iteration: %d, Place: Before Subscribe", i));

                PlcSubscriptionRequest request = connection.subscriptionRequestBuilder()
                        .addChangeOfStateTag("Demo", OpcuaTag.of("ns=2;s=HelloWorld/ScalarTypes/Integer"))
                        .build();

                PlcSubscriptionResponse response = request.execute().get(60, TimeUnit.SECONDS);
                if (response.getResponseCode("Demo") != PlcResponseCode.OK) {
                    throw new RuntimeException("Not OK.");
                }

                printStats(format("Iteration: %d, Place: Before Unsubscribe", i));

                PlcUnsubscriptionRequest unsubscriptionRequest = connection.unsubscriptionRequestBuilder()
                        .addHandles(response.getSubscriptionHandles()).build();
                unsubscriptionRequest.execute();

                printStats(format("Iteration: %d, Place: Before Close", i));
            } finally {
                printStats(format("Iteration: %d, Place: After Close", i));
            }
        }

        printDetails();
    }

    private static void printDetails() {
        System.out.println("CommonPool Parallelism: " + ForkJoinPool.commonPool().getParallelism());
        System.out.println("CommonPool Common Parallelism: " + ForkJoinPool.getCommonPoolParallelism());
        System.out.println("CommonPool Pool Size: " + ForkJoinPool.commonPool().getPoolSize());
        System.out.println("CommonPool Queued Tasks: " + ForkJoinPool.commonPool().getQueuedTaskCount());
        System.out.println("CommonPool Queued Submissions: " + ForkJoinPool.commonPool().getQueuedSubmissionCount());
        System.out.println("CommonPool Active Threads: " + ForkJoinPool.commonPool().getActiveThreadCount());
    }

    private static void printStats(String comment) {
[... 158 lines cut ...]
### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2023-10-10:
Hello @takraj,
It is indeed a bug which will contribute to resource exhaustion. We can introduce dedicated thread (or two) for each OPC UA connection in order to bind these to connection lifecycle.
Your example uses 60s sampling window, could you share what were cycle times you used while tracing this issue? With 60s there is a lot of time for thread context switch even for fairly under-powered devices.

takraj, 2023-10-10:
@splatch I'm not sure what do you mean by 'sampling window'. The example code above does a `connect` -> `subscribe` -> `unsubscribe` -> `disconnect` loop each second. The 60s here is only the maximum amount of time to wait for the subscribtion request to complete.

splatch, 2023-10-10:
I thought of cycle time for subscription. It does influence frequency of "ping" calls made by ua subscriber. I miss interpreted a bit `startSubscriber` block you linked which, but after looking at it second time it, still seems to be a  loop which will occupy a thread and emit `PublisRequest` as long as subscription is active.

There are no doubts on where issue comes from.

PR #1139 (2023-10-13): plc4j-driver-opcua: Fix keepalive threads are never shut down
Attempts to fix #1138 

* OPC-UA keepalive threads now have an exit condition.
* Both keepalive and subscription loops are now executed in their own, dedicated thread, instead of using the `ForkJoin` common pool.
* Keepalive thread now uses the more sensible session timeout, instead of the 10 hour `CONNECTION_LIFETIME`. The keepalive messages are now actually sent, checked with Wireshark.
* Swapped the sleep & communication logics in the keepalive thread.
* Keepalive responses now increment the `senderSequenceNumber`.
* Smoke tests  [... cut]
closed 2023-10-13 by sruehl