apache/plc4x #1174: How to S7-1500 network disconnected but S7HDefaultNettyPlcConnection The connected() method is still true

cptjackwu, 2023-10-26:
### What happened?

An exception occurs when I call the following method，  
 PlcReadResponse response = plcReadRequest.execute().get(10, TimeUnit.SECONDS);

java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcRuntimeException: Disconnected
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
	at com.wh.subscription.S7Client.execute(S7Client.java:392)
	at com.wh.subscription.S7Client.readValues(S7Client.java:328)
	at com.wh.subscription.PlcSubscription.sendRequest(PlcSubscription.java:83)
	at com.wh.subscription.PlcSubscription$WatchdogTimer.lambda$kick$0(PlcSubscription.java:126)
	at com.wh.subscription.PlcSubscription$WatchdogTimer.lambda$runnableNullThrowWrapper$1(PlcSubscription.java:134)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)

I want to use  S7HDefaultNettyPlcConnection isConnected equal to false and then reconnect. This method doesn’t work. What should I do?

 if (this.plcConnection == null || !this.plcConnection.isConnected()) {
                this.plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection(url);
            }

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2023-10-26:
Hi @cptjackwu,
I think this might be related to a problem I observed with the serial transport that I wasn't able to configure the settings for the serial transport as it simply wasn't processed. I know that this once worked, but not quite sure what changed. I'll be investigating this once I find some time.
If you however want to give it a try, I'm happy to help.
Chris

glcj, 2023-10-26:
Hello,

With Siemens PLCs, you can wait a monitoring time of about 6-8 seconds for it to consider the dropped connection (I have tested it on S7-300 and S7-400), so the traffic monitor is set to that time.

In short, you can disconnect the cable for 3 seconds and that method will return that you are connected.

I am going to repeat my tests with version 0.11.0 and the PLCSIM of 1500.

I do not recognize these objects "com.wh.subscription.S7Client.execute(S7Client.java:392)", I imagine they are from your application.

If you enter the connection URL it will help me.

Kind regards,

chrisdutz, 2023-10-26:
And please ignore my mumbling ... my response was intended for another issue that popped up .. sorry for the confusion ;-)