apache/plc4x #1175: How to modify the value inside  TcpTransportConfiguration

cptjackwu, 2023-10-26:
### What would you like to happen?

  default boolean isKeepAlive() {
        return false;
    }

    default boolean isNoDelay() {
        return true;
    }

    default int getConnectTimeout() {
        return 1000;
    }

i want modify connectionTimeout  or isKeepAlive ,but PlcConnection   is no way for me to modify it。

PlcConnection plcConnection = new PlcDriverManager().getConnection(url);

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2023-10-26:
Yikes ... seems I added the reply to this issue to the wrong one ...

I think this might be related to a problem I observed with the serial transport that I wasn't able to configure the settings for the serial transport as it simply wasn't processed. I know that this once worked, but not quite sure what changed. I'll be investigating this once I find some time.

If you however want to give it a try, I'm happy to help.

cptjackwu, 2023-10-27:
I want to modify the default values of these two TCPs. How do I do this?

chrisdutz, 2023-10-27:
Well, as I said: I think this is a vegetal problem with all transport options. Currently I don't know what broke, as this worked before. However I can't say when I'll have the time to have a look.