apache/plc4x #1186: Web Page for "The Connection Cache concept" is outdated as of Release 0.11.0

in-fke, 2023-11-03:
### What happened?

The page https://plc4x.apache.org/users/tools/connection-cache.html no longer reflects the new cache implementation.
And: how do you "dispose" all connections, e.g. for "shutdown" ?

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-02-10:
I've just re-written the entire page .. thanks for reporting. Now the connection-pool page is no longer in the menu.

closed 2024-02-10 by chrisdutz

in-fke, 2024-02-12:
> I've just re-written the entire page .. 

Thanks for following up on this rather quickly.
Question remains: how to do "tear down" a connection pool, e.g. for shutdown / restart mechanisms.
We have a highly configurable framework and may need to "re-deploy" during runtime.

in-fke, 2024-02-12:
> There currently is no such mechanism. Could I ask you to create a Feature-Request for this functionality?

Sure, https://github.com/apache/plc4x/issues/1399