apache/plc4x #1198: [Bug]: 0.11.0 don't support jdk8?

anganing, 2023-11-09:
### What happened?

hello,

the version 0.11.0 dont support jdk 8?

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

splatch, 2023-11-09:
Compile target for recent releases is Java 11. You can try building against older version yourself with `-Djava.version=8`.

chrisdutz, 2023-11-10:
Yeah.. I need Java 11 or newer to build, but Java 8 is the target output to use the compiled libraries.

anganing, 2023-11-10:
> Yeah.. I need Java 11 or newer to build, but Java 8 is the target output to use the compiled libraries.

Does this mean I cannot directly use dependencies from the Maven Central repository? If I want to continue using version 0.10.0, where can I find the corresponding documentation?

chrisdutz, 2023-11-10:
Yes you can continue using the jars from maven central with Java 8... Just it you want to compile plc4x libs yourself, then you need 11.

If you're using the maven central lobby and that's not working, please provide details at that you're using and what the error is, that you're having.

closed 2023-11-10 by anganing