apache/plc4x #1203: [Bug]: S7 driver fails sporadically with null pointer exception

splatch, 2023-11-13:
### What happened?

Observed with released version, I am not quite sure why it fails yet. Registering issue for tracking purposes.

```
java.lang.NullPointerException: null
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.toPlcReadResponse(S7ProtocolLogic.java:302) ~[?:?]
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.read(S7ProtocolLogic.java:294) ~[?:?]
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:152) ~[?:?]
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedRead(BaseOptimizer.java:109) ~[?:?]
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.read(AbstractPlcConnection.java:167) ~[?:?]
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadRequest.execute(DefaultPlcReadRequest.java:55) ~[?:?]
	at org.connectorio.addons.binding.plc4x.handler.task.ReadTask.run(ReadTask.java:66) ~[?:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

### Version

0.11

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2023-11-13:
Perfect ... guess that will be something worth fixing for an upcoming 0.12.0 release ;-) Same with all of our transport-configuration-finetuning stuff ...

glcj, 2024-02-07:
The root cause of this problem was in the driver configuration, which was corrected. So far I have not had this problem and the driver has correctly handled the configuration with additional parameters. If @splatch agrees it can be closed.

chrisdutz, 2024-02-10:
So I guess we can close this issue right? @glcj  and @splatch ?

glcj, 2024-02-10:
+1, close

closed 2024-02-14 by chrisdutz