apache/plc4x #1206: [Bug]: java: 无法访问org.apache.plc4x.java.api.PlcConnection

anganing, 2023-11-15:
### What happened?

hello, 
I use java8 and plc4x 0.11.0.
here is the excption:
[image]
my project cant't use java11 or higher only can use java8, now I wan to use plc4x 0.11.0 what can I do to fix the bug?

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

hongjinlin, 2023-11-15:
Hi, usually this error means that you are trying to run a Java program compiled with a higher version of the compiler on a lower version of the JVM. If you can only use java8 anyway, you can try downloading the source code and compiling your own JAR package locally.

anganing, 2023-11-15:
> Hi, usually this error means that you are trying to run a Java program compiled with a higher version of the compiler on a lower version of the JVM. If you can only use java8 anyway, you can try downloading the source code and compiling your own JAR package locally.

only this way ? can you compiler a java8 jar of plc4x 0.11.0 to maven Central Repository

hongjinlin, 2023-11-15:
> only this way ? can you compiler a java8 jar of plc4x 0.11.0 to maven Central Repository

There is currently no specific plan in place. However, there was a previous discussion on the topic of the Java version, you may want to review it for additional insights.
#1059

chrisdutz, 2023-11-15:
If that's an issue, we really should do a new release soon and ensure the class version is the one of Java 8

chrisdutz, 2023-11-15:
In general we require Java 11 for building plc4x, but it should produce Java 8 compatible libraries. So definitely something we should investigate.

chrisdutz, 2023-12-08:
Ok ... so I guess we need to put in some extra effort to ensure this doesn't happen again. We once had a plugin to ensure this, but that was removed for some reason.

chrisdutz, 2023-12-08:
Ok ... I re-added the animal-sniffer plugin and indeed there are a number of places where we need to fix some things. Sorry for the inconvenience. 

I'll make sure that we will do a new release as soon as possible.

chrisdutz, 2023-12-08:
Ok ... so there was some previous discussion about dropping 1.8 support in the community, however we never really did any form of official vote on it. I'll make sure that we do this. If Java 1.8 support is important to you, I would reccommend that you sign up to the dev@plc4x.apache.org mailinglist (by sending an empty email to dev-subscribe@plc4x.apache.org) However the problem is, that nobody officially admits to be using PLC4X and what they are using it for. So we navigate with what we know and that's basically that only the people working on PLC4X are actually using it.

mrwhy-orig, 2023-12-11:
I had to use a library which was 1.8 only. But lucky me, that was an outdated information and this lib has been updated as well. But as I wrote in the other discussion, for the use case of retrofitting it would make sense to set the compatibility bar as low as possible.

chrisdutz, 2023-12-11:
Well currently the people voting on the dev-list make it appear we're going to be dropping 1.8 as nobody has voter otherwise so far.

mrwhy-orig, 2023-12-11:
I haven't voted ever (and it seems that I haven't got some E-Mails because of bouncers...), as I'm just a user and not a dev of this lib, yet :) It is your decision. I gave my honest opinion on this. If it isn't a big deal to hold on 1.8 compatibilty I would go that way as imho this is a great lib for Retrofitting projects. Maybe the time is not yet there, but the next IT department is just around the corner which will make security update demands / automatic processing data requests to special mechanicel enginerring companies ... So in my view the perfect lib for that use case.

sruehl, 2023-12-13:
As this is open source there is also always the option to create a plc4x-jdk8 fork an maintain it.

mrwhy-orig, 2023-12-13:
@splatch @sruehl yes ofcourse there is always this option, to either fork it or use an outdated version (I tried v9 and it was compatible as well). And of course one should look out for the safest possible way. Don't get me wrong I'm, by no means, saying you have to do it. I'm sorry if I might have made that impression.

sruehl, 2023-12-13:
all good, @mrwhy-orig, don't worry. I think we can consider this ticket now closed as it has been decided to drop java 8 support.

closed 2023-12-13 by sruehl