apache/plc4x #1207: [Bug]: modbus address invalid

anganing, 2023-11-15:
### What happened?

hello,
I read your official documentation that says it's a legitimate address, but it's wrong:
[image]
[image]

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

hongjinlin, 2023-11-15:
You can try 0x00001:BOOL[1]. The Modbus register coil status addresses range from 00001 to 09999, and I just checked, the regular expression matching in the code follows the same pattern.
[ADDRESS_SHORT_PATTERN](https://github.com/apache/plc4x/blob/0d017047bb5d368f58a52a29fe046cf8a1177e5b/plc4j/drivers/modbus/src/main/java/org/apache/plc4x/java/modbus/base/tag/ModbusTagDiscreteInput.java#L32)
[FIXED_DIGIT_MODBUS_PATTERN](https://github.com/apache/plc4x/blob/0d017047bb5d368f58a52a29fe046cf8a1177e5b/plc4j/drivers/modbus/src/main/java/org/apache/plc4x/java/modbus/base/tag/ModbusTag.java#L41)

chrisdutz, 2024-02-10:
Aaaaaa ... now I understand the isssue ... yeah ... I first thought this was related to the "Register 1 = address 0 on the wire" problem ... but yeah ... the address is not in the valid format. I guess it's just missing some zeroes ;-)

closed 2024-03-27 by anganing