apache/plc4x #1212: [Bug]: modbusrtu is not success now

StrawberryBlue, 2023-11-17:
### What happened?

`15:39:33.278 [main] INFO  DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@71bc1ae4
15:39:33.279 [main] INFO  DefaultPlcDriverManager - Registering available drivers...
15:39:33.290 [main] INFO  DefaultPlcDriverManager - Registering driver for Protocol modbus-ascii (Modbus ASCII)
15:39:33.291 [main] INFO  DefaultPlcDriverManager - Registering driver for Protocol modbus-rtu (Modbus RTU)
15:39:33.292 [main] INFO  DefaultPlcDriverManager - Registering driver for Protocol modbus-tcp (Modbus TCP)
15:39:34.211 [main] INFO  Plc4j - Synchronous request ...
15:39:39.719 [main] INFO  NettyChannelFactory - Channel is closed, closing worker Group also
15:39:39.736 [nioEventLoopGroup-2-1] WARN  NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@34a12a3b.
15:39:39.737 [nioEventLoopGroup-2-1] INFO  NioEventLoop - Migrated 1 channel(s) to the new Selector.
15:39:39.739 [nioEventLoopGroup-2-1] WARN  NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@571bf3f7.
15:39:39.739 [nioEventLoopGroup-2-1] INFO  NioEventLoop - Migrated 1 channel(s) to the new Selector.
15:39:39.740 [nioEventLoopGroup-2-1] WARN  NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@14871804.
15:39:39.740 [nioEventLoopGroup-2-1] INFO  NioEventLoop - Migrated 1 channel(s) to the new Selector.
15:39:39.741 [nioEventLoopGroup-2-1] WARN  NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@10d8ef59.`

Now the ModbusRTU cannot run correctly.

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-11-17:
I think the serial transports need a bit of testing ... unfortunately my previous setup seems to be unavailable to me and I'm struggling a bit with testing. 

Could you please give version 0.12.0-SNAPSHOT a try, as we already addressed some configuration issues for serial there.

chrisdutz, 2024-02-06:
Ok .. so I've spotted an error in the Serial Transports and fixed it ... currently looking for a Serial Modbus device to do some further testing, but could you please test the latest SNAPSHOT version?

chrisdutz, 2024-02-10:
Ok ... so I've tweaked quite a bit of stuff around the Serial Transport and Modbus in general ... could you please give this another try and tell us if it's working now?

StrawberryBlue, 2024-02-20:
I tried using the 0.12 version, but it says that the transfer serial is not supported, is there something that has happened to the connection string

StrawberryBlue, 2024-02-20:
MODBUS_RTU_CONNECT_OPTION = "modbus-rtu:serial://COM30?request-timeout=5000&unit-identifier=1&baud-rate=9600&num-data-bits=8&num-stop-bits=1&parity=NO_PARITY";

chrisdutz, 2024-02-20:
If you are using 0.12.0, then please have a look at the online documentation. The parameters for baud-rate etc need a prefix now

StrawberryBlue, 2024-02-20:
sorry,it also exist when i use 

`MODBUS_RTU_CONNECT_OPTION = "modbus-rtu:serial://COM30?request-timeout=5000&unit-identifier=1&serial.baud-rate=9600&serial.num-data-bits=8&serial.num-stop-bits=1&serial.parity=NO_PARITY";`

chrisdutz, 2024-02-20:
What type of device are you trying to communicate with?

StrawberryBlue, 2024-02-21:
I am using a serial device on Windows,
`public static String MODBUS_RTU_CONNECT_OPTION = "modbus-rtu:serial://COM30?request-timeout=5000&unit-identifier=1&serial.baud-rate=9600&serial.num-data-bits=8&serial.num-stop-bits=1&serial.parity=NO_PARITY";
    public static String MODBUS_TCP_CONNECT_OPTION = "modbus-tcp:tcp://127.0.0.1:502";

    @Test
    public void modbusRtuRead(){
        try (PlcConnection connection = PlcDriverManager.getDefault().getConnectionManager().getConnection(MODBUS_RTU_CONNECT_OPTION)) {
            // Check if this connection support reading of data.
            if (!connection.getMetadata().isReadSupported()) {
                log.error("This connection doesn't support reading.");
                return;
            }

            PlcReadRequest.Builder builder = connection.readRequestBuilder();
            builder.addTagAddress("value-1", "400001:UINT[10]");
            PlcReadRequest readRequest = builder.build();

            PlcReadResponse response = readRequest.execute().get(5000, TimeUnit.MILLISECONDS);

            for (String tagName : response.getTagNames()) {
                if(response.getResponseCode(tagName) == PlcResponseCode.OK) {
                    int numValues = response.getNumberOfValues(tagName);
                    // If it's just one element, output just one single line.
                    if(numValues == 1) {
                        log.info("Value[" + tagName + "]: " + response.getObject(tagName));
                    }
                    // If it's more than one element, output each in a single row.
                    else {
                        log.info("Value[" + tagName + "]:");
                        for(int i = 0; i < numValues; i++) {
                            log.info(" - " + response.getObject(tagName, i));
                        }
                    }
                }
                // Something went wrong, to output an error message instead.
                else {
                    log.error("Error[" + tagName + "]: " + response.getResponseCode(tagName).name());
                }
            }

        }catch (Exception e){
            e.printStackTrace();
            log.error("connect fail");
        }
    }`

Like the code , when attempting to establish a connection, the program throws an exception.
`2024-02-21T08:16:30.878+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@2b193f2d
2024-02-21T08:16:30.879+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Registering available drivers...
2024-02-21T08:16:30.884+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Registering driver for Protocol modbus-ascii (Modbus ASCII)
2024-02-21T08:16:30.885+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Registering driver for Protocol modbus-rtu (Modbus RTU)
2024-02-21T08:16:30.886+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Registering driver for Protocol modbus-tcp (Modbus TCP)
2024-02-21T08:16:30.889+08:00  INFO 3276 --- [           main] o.a.plc4x.java.DefaultPlcDriverManager   : Registering driver for Protocol s7 (Siemens S7 (Basic))
org.apache.plc4x.java.api.exceptions.PlcConnectionException: Unsupported transport serial`

StrawberryBlue, 2024-02-22:
I debugged the code and it seems that TcpTransport was used instead of SerialTransport.class, so I manually introduced the plc4j-transport-serial package. At this point, the data can be successfully read once, followed by frequent occurrences of `Select. select() returned accurately 512 times in a row; Rebuilding selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@4954bfd. `Warning, preventing the program from continuing to execute normally.

StrawberryBlue, 2024-02-22:
I continued to check the input logs, and it seems that the previous agent closed the channel after reading it once`Channel is closed, closing worker Group also.`
So I adjusted the code and looped read the data before it was closed. At this point, the data can be read correctly. Thank you for your help.

StrawberryBlue, 2024-02-22:
I continued testing, and when I looped read the data, there was an incomplete message. The test code is as follows
` @Test
    public void modbusRtuRead(){
        try (PlcConnection connection = PlcDriverManager.getDefault().getConnectionManager().getConnection(MODBUS_RTU_CONNECT_OPTION)) {

            while (true) {
                // Check if this connection support reading of data.
                if (!connection.getMetadata().isReadSupported()) {
                    log.error("This connection doesn't support reading.");
                    return;
                }
                PlcReadRequest.Builder builder = connection.readRequestBuilder();
                builder.addTagAddress("value-1", "400001:UINT[100]");
                PlcReadRequest readRequest = builder.build();
                long start = System.currentTimeMillis();
                PlcReadResponse response = null;

                response = readRequest.execute().get(5000, TimeUnit.MILLISECONDS);

                if (response != null) {
                    for (String tagName : response.getTagNames()) {
                        if (response.getResponseCode(tagName) == PlcResponseCode.OK) {
                            int numValues = response.getNumberOfValues(tagName);
                            // If it's just one element, output just one single line.
                            if (numValues == 1) {
                                log.info("Value[" + tagName + "]: " + response.getObject(tagName));
                            }
                            // If it's more than one element, output each in a single row.
                            else {
                                log.info("Value[" + tagName + "]:");
                                for (int i = 0; i < numValues; i++) {
                                    log.info(" - " + response.getObject(tagName, i));
                                }
                            }
                        }
                        // Something went wrong, to output an error message instead.
                        else {
                            log.error("Error[" + tagName + "]: " + response.getResponseCode(tagName).name());
                        }
                    }
                }
                long end = System.currentTimeMillis();
                System.out.println(end - start);
                Thread.sleep(50);
            }
        } catch (Exception e) {
            throw new RuntimeException(e);
        }
    }`

After a period of program execution, there may be a half package situation,

The complete response message should be '0103c8000100000000000000000000000000000000000a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000064a2ba', but the program cannot correctly parse the message due to two returns. Should length verification be added to the modbus return message. The error message is as follows
`2024-02-22 09:41:33.469  WARN 19404 --- [ntLoopGroup-2-1] .p.j.s.GeneratedDriverByteToMessageCodec : Error decoding package with content [0103c8000100000000000000000000000000000000000a00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000]: Index 159 out of bounds for length 159

java.lang.ArrayIndexOutOfBoundsException: Index 159 out of bounds for length 159
	at com.github.jinahya.bit.io.ArrayByteInput.read(ArrayByteInput.java:106) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.DefaultBitInput.read(DefaultBitInput.java:57) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned8(AbstractBitInput.java:63) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned16(AbstractBitInput.java:92) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:135) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:118) ~[bit-io-1.4.3.jar:na]
	at com.github.jinahya.bit.io.AbstractBitInput.readByte(AbstractBitInput.java:106) ~[bit-io-1.4.3.jar:na]
	at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readSignedByte(ReadBufferByteBased.java:388) ~[plc4j-spi-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.spi.generation.ReadBuffer.readByte(ReadBuffer.java:44) ~[plc4j-spi-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readByteArray(ReadBufferByteBased.java:126) ~[plc4j-spi-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.modbus.readwrite.ModbusPDUReadHoldingRegistersResponse.staticParseModbusPDUBuilder(ModbusPDUReadHoldingRegistersResponse.java:112) ~[plc4j-driver-modbus-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.modbus.readwrite.ModbusPDU.staticParse(ModbusPDU.java:181) ~[plc4j-driver-modbus-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.modbus.readwrite.ModbusRtuADU.lambda$0(ModbusRtuADU.java:132) ~[plc4j-driver-modbus-0.12.0.jar:0.12.0]
	at org.apache.plc4x.java.spi.codegen.io.DataReaderComplexDefault.read(DataReaderComplexDefault.java:70) ~[plc4j-spi-0.12.0.jar:0.12.0]
[... cut ...]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562) ~[netty-transport-4.1.94.Final.jar:4.1.94.Final]
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997) ~[netty-common-4.1.94.Final.jar:4.1.94.Final]
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[netty-common-4.1.94.Final.jar:4.1.94.Final]
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30) ~[netty-common-4.1.94.Final.jar:4.1.94.Final]
	at java.base/java.lang.Thread.run(Thread.java:829) ~[na:na]`

chrisdutz, 2024-02-22:
Currently at a conference... Will look into this as soon as I'm back home. And yeah... The serial transport is not included per default. We should probably change that to include all supported transports for a given driver.

StrawberryBlue, 2024-02-28:
Can you reproduce this problem

wenhuancui, 2024-05-27:
Modbus with serial functioning is broken in 0.12.0.

chrisdutz, 2024-05-27:
@StrawberryBlue I just recently updated the Modbus driver to hopefully eliminate the half-packet problem ... could you please give that another try?

StrawberryBlue, 2024-05-28:
it is fixed, great

closed 2024-05-28 by chrisdutz

NdiayeeCheikh, 2025-05-08:
Hi can someone help.
I just have a simple code  for testing the connexion,
but i dont undestand the output prompted:

He's the code :
package com.example.plcapp;
import org.apache.plc4x.java.api.PlcDriverManager;
import org.apache.plc4x.java.api.PlcConnection;
import org.apache.plc4x.java.api.messages.PlcReadRequest;
import org.apache.plc4x.java.api.messages.PlcReadResponse;
import java.util.concurrent.TimeUnit;
public class ModbusClient {

    public static void main(String[] args) {
System.out.println("Bonjour");

        // Adapter le port s..rie selon ton syst..me (ex: COM1 sur Windows,
        String connectionString = "modbus-rtu:serial://dev/ttyV0?request-tim //        System.out.println(plcConnection);
       try (PlcConnection plcConnection = PlcDriverManager.getDefault().getC

        } catch (Exception e) {
                        System.out.println("---------Bonjour---------");
           e.printStackTrace();
        }
    }
}

OUTPUT:
/media/sda4/plc4x/plc-demo # mvn exec:java -Dexec.mainClass="com.example.plc
app.ModbusClient"
[INFO] Scanning for projects...
[WARNING] This build will only read from the build cache, since the clean lifecycle is not part of the build invocation.
[INFO]
[INFO] --------------------< com.example.plcapp:plc-demo >---------------------
[INFO] Building plc-demo 1.0-SNAPSHOT
[INFO]   from pom.xml
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- exec:3.1.0:java (default-cli) @ plc-demo ---
Bonjour
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader org.codehaus.mojo.exec.URLClassLoaderBuilder$ExecJavaClassLoader@9a88a92
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering available drivers...
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-ascii (Modbus ASCII)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-rtu (Modbus RTU)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-tcp (Modbus TCP)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.transport.serial.SerialChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.transport.serial.DefaultSerialTransportConfiguration@39377e46