apache/plc4x #1249: [Bug]: opcua tag read status error

qtvbwfn, 2023-12-06:
### What happened?

UaExpert read OK, plc4x read error

[image]

### Version

V0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2023-12-06:
I think it could be related to string encoding, I recall there was similar issue a while ago. Please have a look on #568.

chrisdutz, 2023-12-06:
But that's in 0.11.0 so it seems you would need to update to v0.11.0

closed 2023-12-07 by qtvbwfn