apache/plc4x #1257: [Bug]: Automatic triggering of a disconnect event after connecting

ibes13, 2023-12-07:
### What happened?

We would like to establish a connection to a S7 plc, tested it with the following plc types
 - S7 1500 (CPU 1515F-2 PN)
 - S7 315

A connection is established for a short time in which we can also read out data from the plc before a DisconnectEvent appears. 

```
2023-12-07 11:21:16,563 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:184)] (executor-thread-2) DisconnectEvent
```
Has anyone ever had this problem ?
The connection to our plc works normally with the Moka7 library

```
try(PlcConnection connection = new DefaultPlcDriverManager().getConnection("s7://"+serverUri+"?remote-rack="+rack+"&remote-slot="+slot);){
}catch(Exception ex){
      ex.printStackTrace();
}
```

```
2023-12-07 11:21:16,164 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:50)] (executor-thread-2) Instantiating new PLC Driver Manager with class loader QuarkusClassLoader:Quarkus Runtime ClassLoader: DEV restart no:13@3f44593a
2023-12-07 11:21:16,164 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:54)] (executor-thread-2) Registering available drivers...
2023-12-07 11:21:16,178 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:61)] (executor-thread-2) Registering driver for Protocol s7 (Siemens S7 (Basic))
2023-12-07 11:21:16,179 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:50)] (executor-thread-2) Instantiating new PLC Driver Manager with class loader QuarkusClassLoader:Quarkus Runtime ClassLoader: DEV restart no:13@3f44593a
2023-12-07 11:21:16,179 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:54)] (executor-thread-2) Registering available drivers...
2023-12-07 11:21:16,180 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:61)] (executor-thread-2) Registering driver for Protocol s7 (Siemens S7 (Basic))
2023-12-07 11:21:16,299 INFO  [org.apa.plc.jav.tra.tcp.TcpChannelFactory org.apache.plc4x.java.transport.tcp.TcpChannelFactory.configureBootstrap(TcpChannelFactory.java:60)] (executor-thread-2) Configuring Bootstrap with Configuration{local-rack=1, local-slot=1, local-tsap=0, remote-rack=0, remote-slot=1, remote-rack2=0, remote-slot2=0, remote-tsap=0, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType=null, readTimeOut=8, ping=false, pingTime=-1, retryTime=4}
2023-12-07 11:21:16,371 INFO  [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.onConnect(S7ProtocolLogic.java:147)] (executor-thread-2) S7 Driver running in ACTIVE mode.
2023-12-07 11:21:16,371 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.onConnect(S7ProtocolLogic.java:148)] (executor-thread-2) Sending COTP Connection Request
2023-12-07 11:21:16,408 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:164)] (executor-thread-2) 11:21:16.408299100  userEventTriggered: S7HMuxImpl#0 Event: org.apache.plc4x.java.spi.events.ConnectEvent@656ba652
2023-12-07 11:21:16,408 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:169)] (executor-thread-2) java.util.NoSuchElementException: watchdog
2023-12-07 11:21:16,462 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$3(S7ProtocolLogic.java:162)] (nioEventLoopGroup-2-1) Got COTP Connection Response
2023-12-07 11:21:16,462 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$3(S7ProtocolLogic.java:163)] (nioEventLoopGroup-2-1) Sending S7 Connection Request
2023-12-07 11:21:16,464 DEBUG [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.channelReadComplete(S7HMuxImpl.java:192)] (nioEventLoopGroup-2-1) 11:21:16.464424800 channelReadComplete: MULTIPLEXOR
2023-12-07 11:21:16,505 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$8(S7ProtocolLogic.java:177)] (nioEventLoopGroup-2-1) Got S7 Connection Response
2023-12-07 11:21:16,506 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$8(S7ProtocolLogic.java:199)] (nioEventLoopGroup-2-1) Sending S7 Identification Request
2023-12-07 11:21:16,511 DEBUG [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.channelReadComplete(S7HMuxImpl.java:192)] (nioEventLoopGroup-2-1) 11:21:16.511971700 channelReadComplete: MULTIPLEXOR
2023-12-07 11:21:16,558 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$15(S7ProtocolLogic.java:213)] (nioEventLoopGroup-2-1) Got S7 Identification Response
2023-12-07 11:21:16,559 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:164)] (nioEventLoopGroup-2-1) 11:21:16.559607300  userEventTriggered: S7HMuxImpl#0 Event: org.apache.plc4x.java.spi.events.ConnectedEvent@29baf6cf
2023-12-07 11:21:16,561 DEBUG [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.channelInactive(S7HMuxImpl.java:198)] (executor-thread-2) channelInactive: S7HMuxImpl#0
2023-12-07 11:21:16,562 DEBUG [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.channelUnregistered(S7HMuxImpl.java:210)] (executor-thread-2) 11:21:16.562598 channelUnregistered: S7HMuxImpl#0
2023-12-07 11:21:16,562 DEBUG [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.channelUnregistered(S7HMuxImpl.java:212)] (executor-thread-2) Unregistered of channel: PRIMARY
2023-12-07 11:21:16,562 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:164)] (executor-thread-2) 11:21:16.562598  userEventTriggered: S7HMuxImpl#0 Event: org.apache.plc4x.java.spi.events.DisconnectEvent@79172d3c
2023-12-07 11:21:16,563 INFO  [org.apa.plc.jav.s7.rea.con.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.connection.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:184)] (executor-thread-2) DisconnectEvent
```

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

glcj, 2023-12-16:
Hello,

Can you try with the SNAPSHOT version of the driver?

I will be attentive to carry out a review,

Greetings,

ibes13, 2023-12-18:
Hello,

I used the snapshot version 0.12.0-SNAPSHOT for the api and the s7 driver.

The DisconnectEvent is still being issued, but the PlcConnection.isConnected() is now always true.

However, we receive the following error when executing a read request.

```
java.util.concurrent.RejectedExecutionException: Task java.util.concurrent.FutureTask@73700b80[Not completed, task = java.util.concurrent.Executors$RunnableAdapter@3d5c822d[Wrapped task = org.apache.plc4x.java.spi.transaction.RequestTransactionManager$TransactionOperation@6f46426d]] rejected from java.util.concurrent.ThreadPoolExecutor@42a48628[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 0]
	at java.base/java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2081)
	at java.base/java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:841)
	at java.base/java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1376)
	at java.base/java.util.concurrent.AbstractExecutorService.submit(AbstractExecutorService.java:123)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.processWorklog(RequestTransactionManager.java:122)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.submit(RequestTransactionManager.java:114)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager$RequestTransaction.submit(RequestTransactionManager.java:195)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.sendInternal(S7ProtocolLogic.java:1424)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.performOrdinaryReadRequest(S7ProtocolLogic.java:1308)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.read(S7ProtocolLogic.java:287)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:152)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedRead(BaseOptimizer.java:109)
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.read(AbstractPlcConnection.java:178)
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadRequest.execute(DefaultPlcReadRequest.java:55)
```

Establishing a connection

```
2023-12-18 10:29:16,987 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:50)] (executor-thread-2) Instantiating new PLC Driver Manager with class loader QuarkusClassLoader:Quarkus Runtime ClassLoader: DEV restart no:1@6f63deee
2023-12-18 10:29:16,987 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:54)] (executor-thread-2) Registering available drivers...
2023-12-18 10:29:16,990 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:61)] (executor-thread-2) Registering driver for Protocol s7 (Siemens S7 (Basic))
2023-12-18 10:29:16,991 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:50)] (executor-thread-2) Instantiating new PLC Driver Manager with class loader QuarkusClassLoader:Quarkus Runtime ClassLoader: DEV restart no:1@6f63deee
2023-12-18 10:29:16,992 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:54)] (executor-thread-2) Registering available drivers...
2023-12-18 10:29:16,993 INFO  [org.apa.plc.jav.DefaultPlcDriverManager org.apache.plc4x.java.DefaultPlcDriverManager.<init>(DefaultPlcDriverManager.java:61)] (executor-thread-2) Registering driver for Protocol s7 (Siemens S7 (Basic))
2023-12-18 10:29:17,020 INFO  [org.apa.plc.jav.tra.tcp.TcpChannelFactory org.apache.plc4x.java.transport.tcp.TcpChannelFactory.configureBootstrap(TcpChannelFactory.java:60)] (executor-thread-2) Configuring Bootstrap with Configuration{local-rack=1, local-slot=1, local-tsap=0, remote-rack=0, remote-slot=1, remote-rack2=0, remote-slot2=0, remote-tsap=0, port=102, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType='null, readTimeOut='0, ping='false, pingTime='0, retryTime='0'}
2023-12-18 10:29:17,026 INFO  [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.onConnect(S7ProtocolLogic.java:147)] (executor-thread-2) S7 Driver running in ACTIVE mode.
2023-12-18 10:29:17,027 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.onConnect(S7ProtocolLogic.java:148)] (executor-thread-2) Sending COTP Connection Request
2023-12-18 10:29:17,027 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:154)] (executor-thread-2) Nombre tuberias: Multiplexor
2023-12-18 10:29:17,027 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:154)] (executor-thread-2) Nombre tuberias: LoggingHandler#0
2023-12-18 10:29:17,028 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:154)] (executor-thread-2) Nombre tuberias: DefaultNettyPlcConnection$1$1#0
2023-12-18 10:29:17,028 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:154)] (executor-thread-2) Nombre tub [... cut ...]
2023-12-18 10:29:17,128 INFO  [org.apa.plc.jav.s7.rea.pro.S7ProtocolEventLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolEventLogic$ObjectProcessor.run(S7ProtocolEventLogic.java:157)] (plc4x-evt-processor-thread-1) ObjectProcessor Bye!
2023-12-18 10:29:17,526 INFO  [org.apa.plc.jav.s7.rea.pro.S7ProtocolEventLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolEventLogic$EventDispatcher.run(S7ProtocolEventLogic.java:233)] (plc4x-evt-dispatcher-thread-1) EventDispatcher Bye!
```

chrisdutz, 2024-02-06:
Could you please share a bit of code that reproduces this problem? I could then test it against my S7-1500 and S7-1200 ...

chrisdutz, 2024-02-10:
Hi,
If you want us to help you with this, we do need a bit more information. If you can't provide us with that, I'll close this issue after the next release (0.12.0)
Chris

ibes13, 2024-02-12:
Hi,
I will test the two scenarios, PLC4X within a normal main method and within a Rest-API, until Wednesday with the current Snapshot and report back the results.
Sebastian

ibes13, 2024-02-14:
Hi,
I have tested connecting and reading with the current snapshot version again.
Reading values now works without an error :blush:

There is still an error when establishing the connection. Also the function "isConnected()" of the interface "PlcConnection" sometimes returns true and sometimes false. However, this does not affect the reading process. 

``` 
2024-02-14 09:26:16,461 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.onConnect(S7ProtocolLogic.java:156)] (executor-thread-2) Sending COTP Connection Request
2024-02-14 09:26:16,466 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: Multiplexor
2024-02-14 09:26:16,470 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: LoggingHandler#0
2024-02-14 09:26:16,474 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: DefaultNettyPlcConnection$1$1#0
2024-02-14 09:26:16,478 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: DefaultNettyPlcConnection$1$2#0
2024-02-14 09:26:16,482 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: CODEC
2024-02-14 09:26:16,485 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: EventListenerMessageCodec#0
2024-02-14 09:26:16,489 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: WRAPPER
2024-02-14 09:26:16,493 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$0(S7ProtocolLogic.java:162)] (executor-thread-2) Nombre tuberias: DefaultChannelPipeline$TailContext#0
2024-02-14 09:26:16,497 DEBUG [org.apa.plc.jav.s7.rea.pro.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.protocol.S7HMuxImpl.encode(S7HMuxImpl.java:130)] (executor-thread-2) ENCODE: PooledUnsafeDirectByteBuf(ridx: 0, widx: 22, cap: 256)
2024-02-14 09:26:16,501 DEBUG [org.apa.plc.jav.s7.rea.pro.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.protocol.S7HMuxImpl.encode(S7HMuxImpl.java:130)] (nioEventLoopGroup-4-1) ENCODE: PooledUnsafeDirectByteBuf(ridx: 0, widx: 22, cap: 22)
2024-02-14 09:26:16,501 INFO  [org.apa.plc.jav.s7.rea.pro.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.protocol.S7HMuxImpl.userEventTriggered(S7HMuxImpl.java:178)] (executor-thread-2) 09:26:16.501457600 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.ConnectEvent@19d4f050
2024-02-14 09:26:16,508 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$3(S7ProtocolLogic.java:176)] (nioEventLoopGroup-4-1) Got COTP Connection Response
2024-02-14 09:26:16,513 DEBUG [org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$3(S7ProtocolLogic.java:177)] (nioEventLoopGroup-4-1) Sending S7 Connection Request
2024-02-14 09:26:16,517 DEBUG [org.apa.plc.jav.s7.rea.pro.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.protocol.S7HMuxImpl.encode(S7HMuxImpl.java:130)] (nioEventLoopGroup-4-1) ENCODE: PooledUnsafeDirectByteBuf(ridx: 0, widx: 25, cap: 256)
2024-02-14 09:26:16,522 DEBUG [org.apa.plc.jav.s7.rea.pro.S7HMuxImpl org.apache.plc4x.java.s7.readwrite.protocol.S7HMuxImpl.encode(S7HMuxImpl.java:130)] (nioEventLoopGroup-4-1) ENCODE: PooledUnsafeDirectByteBuf(ridx: 0, widx: 25, cap: 25)
2024-02-14 09:26:16,526 WARN  [io.net.cha.emb.EmbeddedChannel io.netty.channel.embedded.EmbeddedChannel.recordException(EmbeddedChannel.java:645)] (nioEventLoopGroup-4-1) More than one exception was raised. Will report only the first one and log others.: io.netty.handler.codec.DecoderException: io.netty.handler.codec.EncoderException: MessageToMessageCodec$1 must produce at least one message.
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:98)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
[... cut ...]
public boolean connect(String serverUri, String rack, String slot) {
        try{
            connection = new DefaultPlcDriverManager().getConnection("s7://"+serverUri+"?remote-rack="+rack+"&remote-slot="+slot+"&controller-type=S7_1500");
            Log.info("connected - "+connection.isConnected());

        }catch(Exception ex){
            ex.printStackTrace();
            Log.error("not connected",ex);
        }

        Log.info("s7 client connected ? - "+connection.isConnected());

        return connection.isConnected();
    }
````

closed 2024-02-14 by ibes13