apache/plc4x #1290: [Bug]: TriggeredScraperTask thread silenly completed after TimeoutException

geneross, 2023-12-14:
### What happened?

I've faces with a situation when scheduled earlier TriggeredScraperTask has completed after getting TimeoutException in execution of PlcReadRequest

https://github.com/apache/plc4x/blob/8f914109bb6047af35dc909d3851273e17277b2e/plc4j/tools/scraper/src/main/java/org/apache/plc4x/java/scraper/triggeredscraper/TriggeredScraperTask.java#L126-L135

Changing code to 
```java
                    plcReadResponse = readRequestBuilder
                        .build()
                        .execute()
                        .orTimeout(requestTimeoutMs, TimeUnit.MILLISECONDS)
                        .get();
```
improves the situation. Read request still complete exceptionally and LeasedPlcConnection sets `invalidateConnection` flag so leased connection on close return connection to the connection container with request to invalidate cached connections.  

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

geneross, 2024-01-19:
The cause is in async getting connection. Sometime call of getConnection throws checked exception and  it makes execution thread completion.

geneross, 2024-01-24:
Root the problem is in LeasedPlcConnection. readRequestBuilder() method returns ReadRequestRequest.Builder descendant that later uses result of connection.get() to build instance of PlcReadRequest. But at the time of call leased connection may have already returned and execution will end with NullPoinerException. This in turn cause completion of execution threads in TriggerScraperImpl and kafka source connector no longer retrieve information from server.

Proposed PR here: https://github.com/apache/plc4x/pull/1361/files

chrisdutz, 2024-06-30:
As the proposed PR is merged, can this issue be closed?

closed 2024-08-16 by chrisdutz