apache/plc4x #1291: [Bug]: Can't read array of structures

alegue87, 2023-12-14:
### What happened?

Hello,

i'm trying to read an array of simple structure ( Range type ) in opcua with this query string: **ns=2;s=range;STRING**
This is a view with opcua-expert:

[image]

Following the code the array of length 10 elements is present ( itemCount = 10 ) and the Range class in package org.apache.plc4x.java.opcua.readwrite is created and ready.

[image]

The exception

 _Caused by: org.apache.plc4x.java.api.exceptions.PlcRuntimeException: org.apache.plc4x.java.spi.generation.ParseException: Unsupported case for discriminated type parameters [VariantType=0 arrayLengthSpecified=false]_ 

is raised when plc4x try to instantiate a builder from the identifierType ( not extracted from readBuffer)

[image]

where is the error?

thanks for any reply

### Version

v0.10.0, v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

splatch, 2023-12-19:
While I might be wrong, it looks like combination of several aspects - the custom type combined with (probably) unbounded array. Would you be able to provide a pcap capture with OPC UA packet containing this answer? I would like to see how data is encoded to check if we can detect it in any way and align to base type. The VariantType 0 is null, which is obviously not available in screen you showed.

If you have a possibility to provide prosys simulation server project I'll also be able to repeat it.

splatch, 2024-02-29:
Hello @alegue87, can you post some wireshark capture so we can verify frame decoding? Also, please give a try to PLC4X 0.12 which was released recently.

splatch, 2024-03-27:
Closed due to inactivity and missing input.

closed 2024-03-27 by splatch