apache/plc4x #1346: [Bug]: TriggeredScraperImpl.java should ignore failed tag query.

jywjyw, 2024-01-18:
### What happened?

I'm using TriggeredScraperImpl to fetch data periodically. If any of tag failed, I cannot get the tag which is successful. 
The reason is here: https://github.com/apache/plc4x/blob/v0.11.0/plc4j/tools/scraper/src/main/java/org/apache/plc4x/java/scraper/triggeredscraper/TriggeredScraperImpl.java
    
public static Map<String, Object> convertPlcResponseToMap(PlcReadResponse plcReadResponse) {
        return plcReadResponse.getTagNames().stream()
            .collect(Collectors.toMap(
                name -> name,
                plcReadResponse::getObject
            ));
    }

Should call check all tags are successful before calling plcReadResponse::getObject, or it will throw exception below:
org.apache.plc4x.java.api.exceptions.PlcRuntimeException: Tag 'bool-2' could not be fetched, response was INVALID_ADDRESS
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadResponse.getTagInternal(DefaultPlcReadResponse.java:623)
	at org.apache.plc4x.java.spi.messages.DefaultPlcReadResponse.getObject(DefaultPlcReadResponse.java:111)

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-01-21:
So I guess, if a tag fails, you've got the configuration of your job wrong ... or is there a case where individual tags fail randomly?

chrisdutz, 2024-01-21:
Cause I think in this case the failure is coming from the PLC4X API which simply fails the entire request, if an address is wrong. So changing this for the scraper would be chainging this in the API and I am not sure we want to do that change. I think if someone enters invalid addresses, the entire request should fail.

jywjyw, 2024-01-24:
If a tag is invalid, calling "plcReadResponse.getObject('mytag')" will throw exception below:
```
org.apache.plc4x.java.api.exceptions.PlcRuntimeException: Tag 'bool-2' could not be fetched, response was INVALID_ADDRESS
```
I think the code should be :
```
public static Map<String, Object> convertPlcResponseToMap(PlcReadResponse plcReadResponse) {
        Map<String,Object> ret = new LinkedHashMap<>();
	for (String tagName : plcReadResponse.getTagNames()) {
		if(plcReadResponse.getResponseCode(tagName) == PlcResponseCode.OK) {
			ret.put(tagName, plcReadResponse.getObject(tagName));
		}else {
			 ret.put(tagName, new Exception(plcReadResponse.getResponseCode(tagName).name()));	//should cast it to a special Exception
		}
	}
	return ret;
   }
```

chrisdutz, 2024-06-30:
I think the entire scraper component could use a re-write. Not sure I really like the idea of setting the value to an "Exception".

chrisdutz, 2024-08-02:
So I am still not sure, if we really should even implement this "fix" ... first off all, I don't think it's a bug, it's much more "works as designed". If you put in invalid addresses, the operation fails. I think that's how it should work. What do the others think about this?

chrisdutz, 2024-10-18:
Ok .. so another reason tipped the scale towards changing this ... now only the invalid tags should fail.

closed 2024-10-18 by chrisdutz