apache/plc4x #1362: [Feature Request]: PullRequest is required to reduce server pressure

zhyang-liu, 2024-01-25:
### What would you like to happen?

1. it seems that `SubscriptionRequestBuilder` is not working properly, it aways saying "opcua.tag isn't a PlcSubscriptionTag".
2. It seems that PullRequest is only implemented in MSpec, but is not used.

```golang
{
	subRequest, err := connection.SubscriptionRequestBuilder().
		AddChangeOfStateTagAddress("field1", "ns=1;s=values/int64").
		Build()
	if err != nil {
		fmt.Println(fmt.Errorf("error preparing subscription-request: %s", err.Error()))
		// ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
		// error preparing subscription-request: *opcua.tag isn't a PlcSubscriptionTag

		os.Exit(1)
	}
	subRequestResult := <-subRequest.Execute()
	if subRequestResult.GetErr() != nil {
		fmt.Println(fmt.Errorf("error executing subscription-request: %s", subRequestResult.GetErr().Error()))
		os.Exit(1)
	}
	resp := subRequestResult.GetResponse()
	fmt.Println(resp.String())
	handle, err := resp.GetSubscriptionHandle("field1")
	if err != nil {
		fmt.Println(fmt.Errorf("error getting handle: %s", err.Error()))
		os.Exit(1)
	}
	handle.Register(func(event model.PlcSubscriptionEvent) {
		fmt.Println("got event:\n", event.String())
	})
}

```

### Programming Languages

- [X] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2024-02-06:
Well ... generally the PLC4X API didn't have the concept of a Publish operation, but I have started on working on a PublishAPI extension ... so I guess that#s the reason we haven't got any publish-support yet.

splatch, 2024-10-22:
Please note that golang driver is currently in fairly bad condition. It is not actively maintained and requires huge effort to get functional parity with java version. It misses 1) security handling [encryption & signing] 2) segmentation of messages 3) soon it will also miss support for event subscriptions.

I am closing this issue, as we can't do much about it for now.

closed 2024-10-22 by splatch