apache/plc4x #1363: [Feature Request]: How to use plc4x  build a cpp project

YongningW, 2024-01-25:
### What would you like to happen?

How to use plc4x  build a cpp project? Is the plc4c ready for developers?

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [X] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-01-25:
Well I guess you could probably use PLC4C, but here we only have 3 drivers (PLC4X, Modbus and S7) and all drivers are very limited with respect to features. If using Go is an option, that is very much more advanced and supports a lot more drivers (However the S7 driver is the one that's currently disabled as it needs some refactoring)

YongningW, 2024-02-02:
I wonder when to support  Ethernet/IP protocol in PLC4C ？

chrisdutz, 2024-02-02:
Well ... I would say: As soon as someone implements it or as soon as someone funds the work of someone else implementing it. 

Even if it's possible to port drivers to C, it's still a language most PLC4X contributors rather stay away from.

I could imagine that a RUST implementation would gain more traction, but right now I would not like to give you an ETA on the EIP driver in C nor would I want to give you a guarantee that it will ever happen. (Unless you've got some time, funds or a combination of both)

chrisdutz, 2024-10-18:
Was the initial question answered? If this is a feature request for "plc4cpp" or such or just an informational query? If it was pure informational nature, I'd lean towards considering this issue "closed".