apache/plc4x #1366: [Bug]: Unable to decode Datatype DATE_AND_TIME on S7-300

3lack-eagle, 2024-01-26:
### What happened?

When trying to read a system time of the data type DATE_AND_TIME from our S7-300 controllers, we receive the following exception:
"java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcRuntimeException: Unable to decode Response"

My findings so far:

1. Via another program we query the value with the data type DATE_AND_TIME and get "2024-01-14 22:59:57,685".
2. When querying the system time via your adapter with the data type LTIME, I get the following value: "PT720664H32M27.487843954S". This value seems to me to be an IEC timer that has been running since the beginning of the Epoch timestamp (Thu Jan 01 1970 00:00:01 GMT+0000).

### Version

v0.9.1

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-01-26:
It might be possible that depending on the type of PLC different types need to be used. But unfortunately I don't have an S7 300 available ... possibly Cesar can help with this. Also possibly if you did a WireShark recording of the network traffic and attached that file to this issue, we can provide better support.

glcj, 2024-01-26:
Hello,
I'm going to keep an eye on it.

glcj, 2024-01-29:
Hello,

The fix for the DATE_AND_TIME type is in the following branch

fix/DATE_AND_TIME

I have carried out the tests on an S7-400 and it can read and write correctly.
It would be necessary to verify the writing of the msec.
The modification should not affect any other application, if someone can do a check before pushing the modification to the "develop" branch.

Kind regards,

chrisdutz, 2024-01-29:
I'll try to check out this branch and try if I can still run my manual test against my s7 ... if that works I'm fine with merging it, as the go part is currently disabled anyway.

chrisdutz, 2024-01-29:
Ok ... currently getting failure building PLC4C ... will look into this.

chrisdutz, 2024-01-29:
So it turns out, by switching from simple types to manual types for the dates, I will have to tweak the code generation a bit.

chrisdutz, 2024-01-29:
So please give me a short while to sort this out before merging.

glcj, 2024-01-29:
> So it turns out, by switching from simple types to manual types for the dates, I will have to tweak the code generation a bit.

It is important that the type returned in code generation is always an object and not a native type, I think this could facilitate the insertion of the code for generating the instances. I made a small modification in the code generation which explains the case. Java only.

chrisdutz, 2024-01-30:
Ok ... so I'm implementing an alternate proposal that uses a feature of mspec where a user can define the "encoding" for a field and here provide alternate encodings. I added a "BCD" encoding and am currently testing things with my S7-1500, however it seems that for these devices the response is different (getting 12 byte responses instead of 8 byte ones)

qoomon, 2024-01-30:
Thanks guys for taking care.

chrisdutz, 2024-02-06:
Should be fixed in the next SNAPSHOT ... please verify things work with that.

chrisdutz, 2024-02-06:
And please give us feedback on if this works ... emojis are hard to interpret ;-)

chrisdutz, 2024-02-06:
Oh wow ... great to see that you folks at Amazon are using PLC4X ... at least for S7 ... you might want to consider also using it for some of the other protocols as I see Modbus, ADS and even OPC-UA that we also support. I see you're using Milo for OPC-UA, but Modbus and Ads perhaps instead of re-implementing them, using PLC4X and helping iron out any issues you run into could be a more sustainable approach?

But always happy to see PLC4X being used and us being able to help.

arieleeuw, 2024-02-08:
> Oh wow ... great to see that you folks at Amazon are using PLC4X ... at least for S7 ... you might want to consider also using it for some of the other protocols as I see Modbus, ADS and even OPC-UA [...]

In order to work around this issue and protocol constraints we’ve put some logic on top of the PLC4X library to allow custom handling for specific datatypes and values that go beyond the max pdu of the used controller.

chrisdutz, 2024-02-10:
So are there any other issues still open, that you need to work around? Because, if there are, this would be valuable feedback to us ... and if you know how to support other datatypes, this could be an interesting case for PLC4X to support directly.

But in all cases, if this issue is now solve, I'd like to close the issue and possibly create new ones for other problems.

glcj, 2024-02-10:
Hello,

The DATE_AND_TIME type continues to represent milliseconds as nanoseconds.

**[main] INFO com.ceos.nb.recipes.plc4x_type_dt.Plc4x_Type_DT - DT001 : 1994-01-01T00:17:54.000000032
[main] INFO com.ceos.nb.recipes.plc4x_type_dt.Plc4x_Type_DT - DT002 : 2024-02-10T10:59:16.000000223**

The correct representation should be:

**DT001 : 1994-01-01T00:17:54.032000000
DT002 : 2024-02-10T10:59:16.223000000**

I think that in this case the call to the help function that corrects this problem applies.

I'll do the test and confirm.

Kind regards,

chrisdutz, 2024-02-10:
Will look into this.

chrisdutz, 2024-02-10:
Ok ... indeed ... should be fixed now ... could you please confirm?

glcj, 2024-02-10:
Ok, DATE_AND_TIME, works fine, read/write.

I'm going to check the "DateAndTime" type of asynchronous events.

arieleeuw, 2024-02-12:
Hi Christofer,

Arie here from AWS. Feel free to let me know if you want a quick online meeting in which I can explain which (and why) workarounds and optimizations we have implemented on top of PLC4X in our S7 adapter.

Regards,
Arie

glcj, 2024-02-12:
Hello,

With the latest modification you can read/write date and time types without problem. Final nanosecond presentation was corrected.

I have been able to verify the types ALARM_8 and ALARM_8P, whose timestamp depends on this modification, and they work fine.

It remains to be tested with an S7-300, but I think we can definitely consider closing this point.

Best regards

closed 2024-02-12 by glcj