apache/plc4x #1373: [Bug][plc4j][EIP]: Uneccessary Thread.sleep in EipProtocolLogic

schaebo, 2024-01-31:
### What happened?

There is an unnecessary Thread.sleep in the EipProtocolLogic, which makes the protocol driver unusably slow.

To me it looks like leftover debug code or similar, at least there is no obvious reason to me for it being there.

It is at: https://github.com/apache/plc4x/blob/5cbce4cace5db1b4f2160aff6995deb1fa6bf2ee/plc4j/drivers/eip/src/main/java/org/apache/plc4x/java/eip/base/protocol/EipProtocolLogic.java#L476-L481

### Version

v0.11.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2024-01-31:
I can partly see why it may have been put there, as we are sending a message for each requested tag, this would in some way limit the rate that we send them out. It's a pretty messy way of doing it and 1s is exceptionally long though. I'm happy to merge your PR if @chrisdutz doesn't have anything to add.

Would you be able to share what device you are communicating with? EIP is one of those protocols that seems to have been implemented in a thousand different ways.

splatch, 2024-01-31:
FYI I've implemented decorator over PLC4X 0.8 and 0.11 APIs which throttle requests independently of drivers.
All you need to do is to wrap PLC4X connection into `DecoratorConnection` and pass read decorator (first argument), see example construction:
https://github.com/ConnectorIO/plc4x-extras/blob/extras-0.11.0-connectorio-3/bundles/org.connectorio.plc4x.extras.decorator.throttle/src/test/java/org/connectorio/plc4x/extras/decorator/throttle/ThrottleDecoratorTest.java#L60

I've did it for canopen to keep only given number of SDO calls at the time. This decorator api can be used also to inject artificial locks which do not interfere with driver logic itself.

chrisdutz, 2024-01-31:
Well ... if the driver continues to work without the delay ... Indeed the comment I left indicates that it can be removed ;-)

hutcheb, 2024-02-02:
Thanks for the PR @schaebo :)

closed 2024-02-02 by hutcheb

schaebo, 2024-02-05:
Thanks for the quick response and action.

> Would you be able to share what device you are communicating with?
Most tests I did were with a AB CompactLogix 1769

chrisdutz, 2024-02-05:
I personally have a CompactLogix 1769 (L24ER QBFC1B) as well ... and hopefully will be able to continue my work on enabling Browse functionality as soon as we have the next release out the door.

hutcheb, 2024-02-05:
It should probably be using the message router and connection manager
instead. Not sure why the driver didn’t detect they weren’t available
though.

Are you using the logix driver notation? logix://192.168.1.1