apache/plc4x #1424: [Bug]: Invalid tag item type from old Beckhoff

RuneRoven, 2024-02-25:
### What happened?

Run the read example for GO and changed to a single bool which i can read from other clients on the same computer.
Using an older PLC with Twincat 2 on port 801.

`readRequest, err := connection.GetConnection().ReadRequestBuilder().
		 AddTagAddress("value-bool", "MAIN.myBool").
		 Build()
	 if err != nil {
		 panic(err)
	 }`

`{"level":"debug","connection":"ads.Connection{}","time":"2024-02-25T17:12:04+01:00","message":"created connection, connecting now"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"Connecting"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"connecting"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"Sending request"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"Sending message"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"Reading"}
{"level":"debug","tag":"*model.DirectPlcTag","time":"2024-02-25T17:12:04+01:00","message":"Invalid tag item type"}
{"level":"info","time":"2024-02-25T17:12:04+01:00","message":"Shutting down driver manager"}
{"level":"trace","name":"ads","time":"2024-02-25T17:12:04+01:00","message":"closing driver"}
{"level":"trace","name":"tcp","time":"2024-02-25T17:12:04+01:00","message":"closing transport"}
{"level":"trace","time":"2024-02-25T17:12:04+01:00","message":"Closing"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x28 pc=0x69ea5c]`

### Version

Latest

### Programming Languages

- [ ] plc4j
- [X] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [X] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-02-26:
Would it be possible to test the same on Java?

My reasoning behind this, is that the Java version has been greatly refactored in the past 1,5 years I think the Go version needs some of these updates. It would be good to know, if the Java side works as you expected.

chrisdutz, 2024-02-26:
I rewrote most of the connection establishment something round 1,5 years ago so the java version fetches the symbol-table the data-type table and makes it possible to read complex types. I think the Go version still resolves a symbolic adress but possibly can't read UDTs. It's not a technical problem, it's more a problem of simply not having the time and not being to eager to give one of my former employers something they could need and use ;-) (Not talking about Mapped ... they were awesome)

RuneRoven, 2024-02-26:
I will have another look in the source code to see if i can find the problem. For testing i have a clean plc with a small counter program, so no UDT and only some bool and int

RuneRoven, 2024-02-29:
I have not yet tested the java version. I have however tried the native go ads library and found some errors there. Since i am using an old PLC it returns a symbol list with for example INT16, which in the native library is not handled, it expects INT which make the connection shut down so in the native lib i have added those types and it works. It maybe is the same problem in your code.

chrisdutz, 2024-02-29:
Oh yeah. TwinCat2 is really a topic that we haven't really covered much. Mostly because of the lack of having such a device. If you could possibly start a pr for that, I'd be happy to help.

RuneRoven, 2024-03-04:
I have not had the time to look further into this. I was doing a plugin for benthos to read from ads so I wanted to use the plc4x to have support for more devices, but as for now I ended up with using the go-native-ads library which I had to modify to read TC2.. Most problem was when doing symbol lookup it returned INT16 and UINT16 instead of INT and UINT which caused it to crash. Also for some reason the symbol names need to be in uppercase so I added a conversion in the plugin to change all symbols to uppercase.. I will see if I get the time to look into it again.

chrisdutz, 2026-08-16:
So adding some information on this ... in my work for my commercial ADS driver, I noticed that in general TC2 uses the same protocol as the TC3 does ... however TC2 devices have a number of limitations that require the driver to work differently. So right now TC2 is just not supported by our ADS driver here.