apache/plc4x #1444: Subscribe() tries to cast S7Tag to S7SubscriptionTag and fails [Bug]:

mrwhy-orig, 2024-03-06:
### What happened?

Use the hello-world-plc4x-subscribe example and try to subscribe to a value change.

The subscription leads to following exception:

```
Exception in thread "Thread-0" java.lang.ClassCastException: class org.apache.plc4x.java.s7.readwrite.tag.S7Tag cannot be cast to class org.apache.plc4x.java.s7.readwrite.tag.S7SubscriptionTag (org.apache.plc4x.java.s7.readwrite.tag.S7Tag and org.apache.plc4x.java.s7.readwrite.tag.S7SubscriptionTag are in unnamed module of loader 'app')
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$23(S7ProtocolLogic.java:407)
	at java.base/java.lang.Thread.run(Thread.java:829)

```

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-06-30:
I actually can't confirm this behavior on the current SNAPSHOT version, can it be closed?

chrisdutz, 2024-10-18:
Closing for no response.

closed 2024-10-18 by chrisdutz