apache/plc4x #1450: [Feature Request]: plc4j-driver-modbus need to have Slave Id configuration when reading TagAddress

yangshuyi, 2024-03-13:
### What would you like to happen?

When I try to connect Modbus and read data from address with a slave id like 22, I cannot find the slave id configuration in the plc4j-driver-modbus library. 

I get a little confused, because other's lib like modbus4j has this feature.

Without slave id, it looks always reading from a default slave-id, like 1 which does not exist, then it will cause timeout exception:

java.util.concurrent.ExecutionException: java.util.concurrent.TimeoutException
        at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:396)
        at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2073)
        at com.primustech.siriusplus.dae.modbus.client.ModbusReader.readData(ModbusReader.java:82)
        at com.primustech.siriusplus.dae.modbus.processor.ModbusConnPoolingThread.readData(ModbusConnPoolingThread.java:20)
        at com.primustech.siriusplus.dae.core.processor.ConnPollingThreadAbstract.lambda$run$0(ConnPollingThreadAbstract.java:43)
        at com.primustech.siriusboot.common.base.tracelog.utils.TracelogSpringScheduleUtils.execute(TracelogSpringScheduleUtils.java:20)
        at com.primustech.siriusplus.dae.core.processor.ConnPollingThreadAbstract.run(ConnPollingThreadAbstract.java:42)
        at java.base/java.lang.VirtualThread.run(VirtualThread.java:309)
Caused by: java.util.concurrent.TimeoutException: null
        at org.apache.plc4x.java.spi.netty.NettyHashTimerTimeoutManager.lambda$0(NettyHashTimerTimeoutManager.java:54)
        at io.netty.util.HashedWheelTimer$HashedWheelTimeout.run(HashedWheelTimer.java:715)
        at io.netty.util.concurrent.ImmediateExecutor.execute(ImmediateExecutor.java:34)
        at io.netty.util.HashedWheelTimer$HashedWheelTimeout.expire(HashedWheelTimer.java:703)
        at io.netty.util.HashedWheelTimer$HashedWheelBucket.expireTimeouts(HashedWheelTimer.java:790)
        at io.netty.util.HashedWheelTimer$Worker.run(HashedWheelTimer.java:503)
        at java.base/java.lang.Thread.run(Thread.java:1583)

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-03-13:
The "unit-identifier" doesn't help? https://plc4x.apache.org/users/protocols/modbus.html It's part of the connection-string.

splatch, 2024-03-18:
If its about various slave ids then its a duplicate of #1234.

yangshuyi, 2024-03-21:
Got it. I tried and it works. It might be good if the document can update. Thanks a lot.

closed 2024-03-21 by yangshuyi

chrisdutz, 2024-03-21:
I added the comment "or slave-id" to the description of the unit-identifier.