apache/plc4x #1452: [Bug]: PlcInvalidTagException: [value1] invalid

xp7777, 2024-03-18:
### What happened?

Hello Sir,

I hope this email finds you well. I am writing to seek assistance regarding an issue I encountered while using PLC4X version 0.12.0. Following the examples provided on the official website, I attempted to read the address %I0.2:BOOL. However, I encountered an error. Could you please advise on the possible reasons for this error?

Here is the code:
package com.yxp;

import org.apache.plc4x.java.api.PlcConnection;
import org.apache.plc4x.java.api.PlcDriverManager;
import org.apache.plc4x.java.api.messages.PlcReadRequest;
import org.apache.plc4x.java.api.messages.PlcReadResponse;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.util.concurrent.TimeUnit;

public class plc {

    private static final Logger logger = LoggerFactory.getLogger(plc.class);

    private static void read() {
        String connectionString = "s7://192.168.1.181";

        try (PlcConnection plcConnection = PlcDriverManager.getDefault().getConnectionManager().getConnection(connectionString)) {

            // Check if this connection support reading of data.
            if (!plcConnection.getMetadata().isReadSupported()) {
                logger.error("This connection doesn't support reading.");
                return;
            }

            // Create a new read request:
// - Give the single item requested an alias name
            PlcReadRequest.Builder builder = plcConnection.readRequestBuilder();

            builder.addTagAddress("value1","%I0.2:BOOL");

            PlcReadRequest readRequest = builder.build();
            PlcReadResponse response  = readRequest.execute().get(5000, TimeUnit.MILLISECONDS);
            System.out.println( response.getTagNames()+"---"+response.getResponseCode(response.getTagNames().toString()));

        }catch (Exception e){
            e.printStackTrace();

        }

    }

    public static void main(String[] args) {
        read();
    }
}

Error message as follows：
io.netty.handler.codec.DecoderException: io.netty.handler.codec.EncoderException: MessageToMessageCodec$1 must produce at least one message.
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:98)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:346)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:318)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
[... 84 lines cut ...]
### Version

0.12.0

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-03-18:
When running your example, I only get issues with the last line:

After changing:

```
        System.out.println( response.getTagNames()+"---"+response.getResponseCode(response.getTagNames().toString()));
```

To:

```
       for (String tagName : response.getTagNames()) {
                System.out.println( tagName+"---"+response.getResponseCode(tagName));
            }
```

All worked fine on my machine ... what type of device are you connecting to?

xp7777, 2024-03-18:
thank you sir!
"I'm using TIA v16 to simulate PLC operation on my computer."

"You're welcome! I'm glad your issue is resolved. It indeed seems to have been caused by the problem with the for loop iteration."

chrisdutz, 2024-03-18:
I think the TIA Simulator only has Network support in it's "Advanced" version. Are you using that?

chrisdutz, 2024-03-18:
Happy to help ... would you mind explain a bit what you are working on?

wenzi121, 2024-03-28:
i have the same problem on version 0.12.0,but it is normal in version 0.11.0.
[image]

xp7777, 2024-04-03:
> 我认为 TIA 仿真器仅在其“高级”版本中具有网络支持。你在用那个吗？

I am using Nettoplcsim S7o software, which allows me to connect to TIA PLCSIM. I am developing an  application using JavaFX that can read PLC data and store it in a database.

closed 2024-04-03 by xp7777

chrisdutz, 2024-04-03:
This is a different issue as the one reported here, right? Would you mind opening a new issue (Ideally attaching a wireshark recording of your S7 traffic?