apache/plc4x #1457: [Bug]: plc4j modbus will be blocked when trying to read value from address

yangshuyi, 2024-03-21:
### What happened?

When I try to get the value from one Modbus address, the program will be blocked.

code:
            PlcReadRequest.Builder plcBuilder = plcConn.readRequestBuilder();
            plcBuilder.addTagAddress(point.getId(), tagAddress);
            PlcReadRequest request = plcBuilder.build();
            PlcReadResponse response = request.execute().get();
            LOGGER.debug("Get response");

the log will never print out.

1. This issue will happen after running the program for a long time, maybe after one hour.
2. I use new DefaultPlcDriverManager().getConnection(serverUrl); to create modus connection.
3. I create multiple connection instances for the same Modbus server with different slaveId (more than 10) for different value readers.

Is there anyway to set read timeout?

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-03-21:
Unfortunately you left away most of the important bits.
We generally have 3 flavors of modbus:
- modbus-tcp
- modbus-rtu
- modbus-ascii
Could you please tell us the connection string and the resource addresses?

Also, as I'm trying to get more feedback on where PLC4X is being used and what for, It would be great if you could also leave a bit of information on your usecase.

chrisdutz, 2024-03-21:
Ok ... that simplifies things ... 

I guess what would help, would be if you created a wireshark recording that records this problem and could attach that to this issue.

yangshuyi, 2024-03-22:
this should because poor network. Once get reading timeout, instead of reuse the connection for next query, need to close the connection and re-connect again.

closed 2024-03-22 by yangshuyi