apache/plc4x #1469: [Bug]: PLC4Py can not read from virtual Modbuspal with Port 5020

ottlukas, 2024-03-26:
### What happened?

Clean Xubuntu install.
Installed Python 3.11.6

Setup PLC4Py from source using the following source code:
```
import asyncio
from plc4py.PlcDriverManager import PlcDriverManager

connection_string = "modbus://127.0.0.1:5020"
driver_manager = PlcDriverManager()

async def communicate_with_plc():
    async with driver_manager.connection(connection_string) as connection:
        with connection.read_request_builder() as builder:
            builder.add_item("Random Tag", "4x00001[10]")
            request = builder.build()

    future = connection.execute(request)
    await future
    response = future.result()

asyncio.run(communicate_with_plc())
```
result = PlcResponse(response_code=<PlcResponseCode.NOT_CONNECTED: 12>)

Attached Wireshark Capture: 
[plc4py_modbuspal_port5020.zip](https://github.com/apache/plc4x/files/14757528/plc4py_modbuspal_port5020.zip)

[PLC4Py_ModpusPal_Port5020.txt](https://github.com/apache/plc4x/files/14757553/PLC4Py_ModpusPal_Port5020.txt)

### Version

v0.12.0

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

ottlukas, 2024-03-27:
```
ERROR:asyncio:Exception in callback _SelectorSocketTransport._call_connection_lost(None)
handle: <Handle _SelectorSocketTransport._call_connection_lost(None)>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 1117, in _call_connection_lost
    super()._call_connection_lost(exc)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 896, in _call_connection_lost
    self._protocol.connection_lost(exc)
  File "/home/luk/Develop/plc4x/plc4py/.venv/lib/python3.11/site-packages/plc4py/spi/Plc4xBaseProtocol.py", line 38, in connection_lost
    raise ConnectionError
ConnectionError
```

hutcheb, 2024-04-09:
In the connect_lost method of the protocol an exception was always raised. This method is also used during the close process, however with the exc parameter set to None. It now only raises a ConnectionException if connection_lost is called with an exception.

Can you please try it again?

ottlukas, 2024-04-10:
Yes thank you very much it is working. Besides this error:
```
DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "/usr/lib/python3.11/runpy.py", line 198, in _run_module_as_main
    return _run_code(code, main_globals, None,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/runpy.py", line 88, in _run_code
    exec(code, run_globals)
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39, in <module>
    cli.main()
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430, in main
    run()
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284, in run_file
    runpy.run_path(target, run_name="__main__")
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 321, in run_path
    return _run_module_code(code, init_globals, run_name,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 135, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/home/luk/.vscode/extensions/ms-python.debugpy-2024.4.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124, in _run_code
    exec(code, run_globals)
  File "/home/luk/Develop/main.py", line 20, in <module>
    asyncio.run(communicate_with_plc())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/luk/Develop/main.py", line 16, in communicate_with_plc
    response = future.result()
               ^^^^^^^^^^^^^
AttributeError: 'coroutine' object has no attribute 'result'
```
Needed to change it to:
```
import asyncio
from plc4py.PlcDriverManager import PlcDriverManager

connection_string = "modbus://127.0.0.1:5020"
driver_manager = PlcDriverManager()

try:
  async def communicate_with_plc():
    async with driver_manager.connection(connection_string) as connection:
        with connection.read_request_builder() as builder:
            builder.add_item("Random Tag", "4x00001[11]")
            request = builder.build()

        future = connection.execute(request)
        response = await future

        # Access and process data from the response here
        data_value = response.response_code  # Assuming "Random Tag" is a key
        print("Data:", data_value)

except Exception as e:
  print("Error communicating with PLC:", e)

asyncio.run(communicate_with_plc())
```
Response:
```
INFO:root:Instantiating new PLC Driver Manager with class loader <pluggy._manager.PluginManager object at 0x76dc9e31a090>
INFO:root:Registering available drivers...
INFO:root:... umas .. OK
INFO:root:... modbus .. OK
INFO:root:... mock .. OK
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:root:Sending read request to Modbus Device
DEBUG:root:Reading tag ModbusTagHoldingRegister(address=0) from Modbus Device
Data: PlcResponseCode.OK
```

closed 2024-04-10 by ottlukas

hutcheb, 2024-04-10:
Yeah, take a look at the updated example on the website. I changed it a bit.