apache/plc4x #1472: [Bug]: opcua unsupported transport tcp

alexiskhoury, 2024-04-01:
### What happened?

Unable to connect to opcua server, the connection aborts with the following message:

`Exception in thread "main" org.apache.plc4x.java.api.exceptions.PlcConnectionException: Unsupported transport tcp
        at org.apache.plc4x.java.spi.connection.GeneratedDriverBase.getConnection(GeneratedDriverBase.java:340)
        at org.apache.plc4x.java.spi.connection.GeneratedDriverBase.getConnection(GeneratedDriverBase.java:293)
        at org.apache.plc4x.java.DefaultPlcDriverManager.getConnection(DefaultPlcDriverManager.java:79)`

Tried to add plc4j-transport-tcp dependency with no avail. Tried also to use v0.11.0 and got the same error. I also tried to change the connection string from opcua:tcp to opc.tcp but got a driver registration issue.
Please this is an urgent matter, your help is appreciated.

P.S.: Tested against the freely available opc server "opc.tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer", hence my connection string is : "opcua:tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer"

Alexis

### Version

v.0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [X] OPC-UA
- [ ] S7

chrisdutz, 2024-04-01:
Try to omit the ":tcp" part...

alexiskhoury, 2024-04-01:
Omitting the :tcp doesn't work, got the same error.

chrisdutz, 2024-04-01:
The name of the driver is opcua not opc (https://github.com/apache/plc4x/blob/7ec9bdc56aa06d0a4448b190b89b869ff8e853ad/plc4j/drivers/opcua/src/main/java/org/apache/plc4x/java/opcua/OpcuaPlcDriver.java#L44)

alexiskhoury, 2024-04-01:
I am using opcua in my connection string as mentioned in my first post and this doesn't work. Any hint?

chrisdutz, 2024-04-02:
Ok ... so I tried your connection string and am seeing the same issue. I would assume that something is going wrong during the connection process. Unfortunately I am intentionally staying a bit away from OPC-UA as the rest of the community usually has this base covered. However I could imagine that most people here had some days off during the easter holidays, so please be patient for them to recover from that ... I am sure someone will be able to help you.

@ceos01 @hutcheb ... could you please have a look?

alexiskhoury, 2024-04-02:
Hi Christofer,

Thanks for your cooperation. I have seen other issues related to opc ua dealing with stale mode / tokens https://github.com/apache/plc4x/issues/1364 still not fully resolved. There is no rush to get these issues sorted out, specially the connection issue that I have mentionned and which seems to be also an issue in v.0.12.0  as stated in the link provided above. I am promoting plc4j as a multi-protocol gateway in our company and we want to succeed witht this project. OPC UA connectivity is a must given that most of our third party machine providers use this protocol.

Alexis

chrisdutz, 2024-04-02:
I guess the main problem is, that nobody working on PLC4X is currently being paid to do that ... so we're all investing our free time. I hope that's gonna change back some day, as the project was progressing a lot faster as we still had full-time contributors. Unfortunately the industry is not accustomed to supporting open-source development, so my hopes are not too high.

alexiskhoury, 2024-04-02:
This is a very promising project and hope that the community will pursue the development process. Plenty of IoT applications will rely on your PLC4X driver to connect PLCs/Machines at factory level to the Internet. Although, as said before, your multiprotocol feature makes PLC4X an efficient central gateway. 

I had a response from Lukasz from the development team and apparently there might be a clue for the problem I am facing. As per Lukasz statement, "Also note that http transport/xml encoding is unsupported. We cover only binary marshaling".
This might be the issue, I will try the driver with a local opc demo server to see if the problem persists and revert to you guys.

splatch, 2024-04-02:
Its probably a missing dependency, can you add `plc4j-transport-tcp` to your classpath?

splatch, 2024-04-02:
> Thanks for your cooperation. I have seen other issues related to opc ua dealing with stale mode / tokens #1364 still not fully resolved. There is no rush to get these issues sorted out, specially [...]

To throw some context in above issue - I've found cause for missing subscriptions, however I've also found a lower level issue related to renewing of OPC-UA security token which I couldn't fix yet. I have to book for that at least two-three days to understand logic which we should build, and test it with software and hardware based servers.

alexiskhoury, 2024-04-02:
> Its probably a missing dependency, can you add `plc4j-transport-tcp` to your classpath?

I am using maven to manage dependencies, and when `plc4j-transport-tcp` is added to the pom.xml, this doesn't solve the issue.

alexiskhoury, 2024-04-02:
Hi,

I am afraid that opcua driver is not working whatever I do. I have tested several opc ua servers locally, some are java based, others are based on pyhton and C and none of them worked. Although, as mentioned above, adding the plc4j-transport-tcp to pom.xml doesn't change anything. One of the servers I am using can be reached on github https://github.com/digitalpetri/opc-ua-demo-server and can be setup quickly on any platform (if you would like to check).

splatch, 2024-04-02:
@alexiskhoury I made an attempt to run 0.12 and 0.13-SNAPSHOT (current develop) wit prosys simulation server, both looks fine. For me root cause of troubles (why it didn't work in first run) was extra trailing slash after `OPCUA/SimulationServer`. The endpoint matcher is strict, so it looks for exact match.
Can you share log or strack trace you get?

Example program for 0.12+ releases:
```
        String keystoreParams = "discovery=true"
            + "&application-uri=urn:apache:plc4x:client"
            + "&negotiation-timeout=60000"
            + "&security-policy=NONE"
            + "&message-security=NONE";
        try (PlcConnection connection = new DefaultPlcDriverManager().getConnection("opcua:tcp://arch:4840/OPCUA/SimulationServer?" + keystoreParams)) {
            CompletableFuture<? extends PlcSubscriptionResponse> future = connection.subscriptionRequestBuilder()
                .addChangeOfStateTagAddress("c", "ns=0;i=2258")
                .build().execute();
            future.whenComplete((r, e) -> {
                if (e != null) {
                    e.printStackTrace();
                    return;
                }
                r.getSubscriptionHandle("c").register(event -> {
                    for (String field : event.getTagNames()) {
                        System.out.println(field + " " + event.getObject(field) + " " + event.getResponseCode(field));
                    }
                });
            }).get();
            Thread.sleep(60_000);
        } catch (PlcConnectionException e) {
            throw new PlcRuntimeException(e);
        } catch (Exception e) {
            e.printStackTrace();
        }
```

alexiskhoury, 2024-04-03:
Many thanks for your help,

I started from scratch with the code provided by @splatch the connection didn't work first. When I added `plc4j-transport-tcp` to my pom.xml, the code worked. For the issue I mentioned, I presume that there was a problem with my IDE configuration, potentially with the classpath.

Feature Request.: Do you have any plan to develop IE61850 protocol? This is will be awesome and very useful for utility scale power plants

closed 2024-04-03 by alexiskhoury

chrisdutz, 2024-04-04:
Please create a new Feature request ... I'll respond there.