apache/plc4x #1476: [Bug]:  Issue with Thread Loop Execution When Attempting Connection to Non-PLC Device

xp7777, 2024-04-03:
### What happened?

hello sir!

When attempting to connect my computer, the following threads are looping in retries:

14:40:00.335 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
14:40:00.345 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@1eecea74
14:40:01.335 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Reconnecting primary channel.
14:40:01.336 [nioEventLoopGroup-14-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
14:40:01.336 [nioEventLoopGroup-14-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@7a9b5a44
14:40:02.334 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
14:40:02.343 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@1eecea74
14:40:03.335 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Reconnecting primary channel.
14:40:03.336 [nioEventLoopGroup-15-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
14:40:03.336 [nioEventLoopGroup-15-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@7a9b5a44
14:40:04.335 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.

Is there an API available to terminate this connection, considering that the target is not an actual PLC but rather the IP address of my computer?

### Version

0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

xp7777, 2024-04-07:
> Hello, You can check the documentation [1] and examples in the repositories.
> 
> Additionally, you must document the failure including the connection URL.
> 
> 1. https://plc4x.apache.org/users/protocols/s7.html

sir， That means the IP in the URL must be that of the PLC. Otherwise, if the IP can be pinged but is not that of a PLC, the connection program will not stop.

The following is the log content：
10:01:42.674 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
10:01:42.683 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@3272d538
10:01:42.684 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
10:01:42.686 [nioEventLoopGroup-89-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
10:01:42.686 [nioEventLoopGroup-89-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - 10:01:42.686604900 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.DisconnectedEvent@2aafdb9
10:01:42.686 [nioEventLoopGroup-89-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@4a9e6cf7
10:01:42.688 [pool-8-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - Timeout during Connection establishing, closing channel...
10:01:43.675 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
10:01:43.684 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@3272d538
10:01:43.685 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
10:01:43.687 [pool-8-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - Timeout during Connection establishing, closing channel...
10:01:43.687 [nioEventLoopGroup-90-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
10:01:43.687 [nioEventLoopGroup-90-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - 10:01:43.687140900 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.DisconnectedEvent@4b5276d5
10:01:43.687 [nioEventLoopGroup-90-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@4a9e6cf7
10:01:44.674 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
10:01:44.683 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@3272d538
10:01:44.684 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
10:01:44.686 [nioEventLoopGroup-91-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
10:01:44.686 [nioEventLoopGroup-91-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - 10:01:44.686527300 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.DisconnectedEvent@7ed5885a
10:01:44.686 [nioEventLoopGroup-91-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@4a9e6cf7
10:01:44.688 [pool-8-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - Timeout during Connection establishing, closing channel...
10:01:45.674 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.p.S7HPlcConnection - Creating prymary connection.
10:01:45.683 [plc4x-s7ha-thread-1] INFO  o.a.p.j.t.tcp.TcpChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.s7.readwrite.configuration.S7TcpTransportConfiguration@3272d538
10:01:45.684 [plc4x-s7ha-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
10:01:45.687 [nioEventLoopGroup-92-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - Unregistered of channel: PRIMARY
10:01:45.687 [nioEventLoopGroup-92-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - 10:01:45.687127300 userEventTriggered: Multiplexor Event: org.apache.plc4x.java.spi.events.DisconnectedEvent@53f849b2
10:01:45.687 [nioEventLoopGroup-92-1] INFO  o.a.p.j.s.r.protocol.S7HMuxImpl - io.netty.channel.embedded.EmbeddedEventLoop@4a9e6cf7
10:01:45.688 [pool-8-thread-1] INFO  o.a.p.j.s.r.protocol.S7ProtocolLogic - Timeout during Connection establishing, closing channel...

chrisdutz, 2024-04-11:
I wouldn't say "not planned" I'll definitely look into this... Just not immediately

chrisdutz, 2024-04-12:
So I just tried connecting to an S7 on an IP address, that doesn't exist and the connection simply failed with "connection is not possible". So I can't confirm this loop that you are referring to. I both tried connecting to an existing IP that is not an S7 as well as a completely not existing. Both correctly report not being able to connect.

closed 2024-04-12 by chrisdutz