apache/plc4x #1482: IEC-61850

alexiskhoury, 2024-04-04:
### What would you like to happen?

Hi,

Thanks again for this great piece of software. I hope that you can develop the IEC-61850 protocol which is an electrical utility protocol used in substations and IEDs (Intelligent Equipment Devices). This protocol is often used in renewable power plants even at machine level for data exchange. The GOOSE message feature allows fast tripping and is also used for control purposes.

I believe that if PLC4X implements this protocol, it will serve plenty of applications mainly in the energy sector.

Regards,
Alexis

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-04-04:
We already have an implementation of iec-60870 which also seems to come from that angle ... but I guess the thing we would need is: 
- Access to the specification
- Access to some demo equipment to play with
- Time to do that
In the past when such requests came up there was usually some company having an advanced interest in this happening and usually provided such ... ideally also some financing of the development.

Unfortunately there is plenty of interest in us implementing all sorts of protocol adapters, however the willingness to acknowledge and reward our activies, which usually happen during our free time, is almost not existent.

I even tried crowdfunding development efforts, but that too didn't get a single initiative to get even one Euro in support.

So personally, I'd be happy wo work on something like that ... but I won't do it for free in my free time and even spend money for purchassing specs and equipment.

alexiskhoury, 2024-04-06:
Sad to hear this news and somehow I am frustrated given that PLC4X is already in use in several projects like apache Camel, apache Nifi, apache IoTDB without forgetting other platforms widely in use like Kafka and extensively using PLC4X.
Hope that you will get the support you need very soon to keep PLC4X up-to-date.

closed 2024-04-06 by alexiskhoury