apache/plc4x #1507: [Feature Request]:

nsgfx, 2024-04-10:
### What would you like to happen?

I want to connect s7-200 smart,and I also add "?controller-type=S7_200" in connect string.But it doesn't work。What cant I do?

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2024-04-11:
You could attach a wireshark recording to this issue, so we can investigate. With the information you provided, I would say: "It should work" ... for more insight we would need the capture.

chrisdutz, 2024-04-18:
I think @ceos01 is probably the only one of us who could have access to such a device. So I guess if he doesn't implement it, that's not going to happen any time soon. 

But having a look at the history:
On 26.12.22 Jinlin Hong added that in commit 3f46e01c49abce7d21c295ec4c14e3d1f10c2a43 with the comment "add s7-200 (Smart) type support" and on 27.04.2024 Cesar removed that again in commit 89f44d97e94b28b8e2b1f0ffb140fae2a9dfe86d with the comment "Feat/s7ha"

So ... @ceos01 ... is there a reason you removed that again? If removing this breaks things, we should probably find a way to allow both.

glcj, 2024-04-18:
Uhh,
It's probably a mistake on my part.
I have not tested the driver with an S7-200. That change can be reverted, but I don't have an S7-200 for testing. Well, I'll check in the PG if the emulator supports the connection.
Excuse me,

nsgfx, 2024-04-19:
Well,when I change the code same as https://github.com/apache/plc4x/commit/3f46e01c49abce7d21c295ec4c14e3d1f10c2a43, it can work.But I find a  bug,that is the lib generate Destination TSAP: 0101,when I am connecting plc-200 by STEP 7-MicroWIN SMART, I can not connect by java code. It may change the   Destination TSAP to 0301,it may work better.

chrisdutz, 2024-04-19:
Well when having a look at the S7TsapIdEncoder, you would need to set the device type to "OTHER" and  rack to 0 and slot to 1 to get that tsapid.

But ... looking into this a bit more, yeah ... the Device-Type is currently hard-coded ... should be easy to make this a configuration option.

chrisdutz, 2024-04-19:
I just noticed, that we already have an option to do that as we have the "local-tsap" and "remote-tsap" ... so in your case setting remote-tsap to the value you want (keep in mind to convert the hex value into an integer) ... but I'll also add the device-group config options.

nsgfx, 2024-04-19:
Well,seems all right.Thanks

chrisdutz, 2024-04-19:
Ok ... my update is in ... now it has to pass through the CI to become visible.