apache/plc4x #1509: [Bug]: [ADS] Writing a List of Struct writes nothing

IsmoLeszczynski, 2024-04-11:
### What happened?

The PLC I'm developing against has a symbol of type List that contains a list of Structs, which aren't exposed as their own symbols (at least not included in the browse results when retrieving "symbols"). Reading the current value from the symbol produces the correct List of Struct results, but when I try to write an updated value back to it, the driver resolves it to an empty byte[]. This seems to be happening within `AdsProtocolLogic.serializeInternal()`, when iterating the list entries it determines each entry to be a list as well, since the same dataType is used for the child iteration as well, and the `DataItem.staticSerialize()` can't handle that.

How the tag is seen in TwinCAT:
[image]

The browsed symbol:
[image]

The attempted write value:
[image]

### Version

v0.12.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [X] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-04-11:
Could you add wireshark recordings of this? I only have 4h of work on PLC4X per week and I would want to make it as efficient as possible.

IsmoLeszczynski, 2024-04-12:
Sure, here you go:
[ads-invalid-write-list-of-struct.zip](https://github.com/apache/plc4x/files/14956297/ads-invalid-write-list-of-struct.zip)

The capture contains the connection, browsing of symbols, reading the List-of-Struct symbol and writing back to it - the data bit in the write is completely empty.

I also created a PR(#1517) that fixed my use case, the issue seemed to only be that for Lists the children were also processed as Lists.

IsmoLeszczynski, 2024-04-12:
And I apparently closed this by mistake, my bad.

IsmoLeszczynski, 2024-04-12:
Closing this as the issue has been fixed with the merge of #1517

closed 2024-04-12 by IsmoLeszczynski