apache/plc4x #1519: [Bug]: ADS Driver doesn't seem to support connections via VPN

chrisdutz, 2024-04-12:
### What happened?

When trying to connect to an ADS device via a VPN connection, the remote device seems to simply hang up.

### Version

v0.13.0-SNAPSHOT

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [X] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

IsmoLeszczynski, 2024-04-12:
Just as additional information relating to this, I've been working on a solution and testing it via an OpenVPN connection successfully, so some VPN connections work.

What I did have issues with was connecting via IP instead of hostname, don't know that's related - using IP caused the PLC to terminate the connection immediately, but using hostname worked fine. This was a bit puzzling, but I haven't used ADS enough to say is this as it should be.

chrisdutz, 2024-04-12:
I recently setup my test-environment to be only reachable via VPN (Unify Gateway) ... so when using the built-in MAC VPN (L2TP over IPSec) and using an IP I was having the remote simply hang up ... if this is just a documentation problem, then I would be super happy ;-)

chrisdutz, 2024-04-12:
I think the problem might be related to my device getting an IP address that is generally not reachable from the ADS device. So the device has the IP 192.168.23.20 and a netmask for 255.255.255.0 ... my local dialin computer has an address of 192.168.2.6 ... I would probably try to configure the VPN to give my client an IP on the remote network instead of using the VPN tunnel as default gateway.