apache/plc4x #1529: [Bug]:

hyzable, 2024-04-15:
### What happened?

hi:
    There is a mistake happend when i run command "mvn -P with-c install" in win10.
    I'm not sure if this is a bug.
```
INFO] --- cmake:3.26.3-b1:generate (cmake-generate-compile) @ plc4c ---
[INFO] Downloading binaries to D:\apache-plc4x-0.12.0\plc4c\target\dependency\cmake
[INFO] Configured Artifact: com.googlecode.cmake-maven-project:cmake-binaries:windows-x86_64:3.26.3-b1:jar
[INFO] cmake-binaries-3.26.3-b1-windows-x86_64.jar already unpacked.
[INFO] Executing D:\apache-plc4x-0.12.0\plc4c\target\dependency\cmake\bin\cmake.exe
-- The C compiler identification is MSVC 19.29.30145.0
-- The CXX compiler identification is MSVC 19.29.30145.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - broken
CMake Error at target/dependency/cmake/share/cmake-3.26/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/apache-plc4x-0.12.0/plc4c/target/build/CMakeFiles/CMakeScratch/TryCompile-9o5nae
```

### Version

v0.12

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [X] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-04-15:
It seems there's something fishy with your compiler ... as far as I understood this message happens when CMake tries to compile a dummy test project in order to check if all works. Have you setup your system according to our guides here: https://plc4x.apache.org/developers/preparing/windows.html?

hyzable, 2024-04-15:
Excpet for WinBuilds.
The website (https://win-builds.org/) could not be visited.

chrisdutz, 2024-04-15:
Grmpf ... these windows build projects are a bit like fungi gowing in some petri-dish ... you never know how they look the next time you have a look ... I'll try to remember having a look this friday, when I have my next slot of PLC4X-Time (Or hopefully someone else beats me to it)

chrisdutz, 2024-04-26:
Ok .... so I've tried updating my documentation for building PLC4X on windows based on my Windows VM, however the cmake plugin currently seems to have issues building on Aarch64 systems (Windows VMs on Arm Macs) ... will try the same on my Intel device.

chrisdutz, 2024-04-26:
From what I see at the official cmake page (https://cmake.org/download/), the following versions should probably need to be deployed:
- windows-arm_64
- windows-x86_64
- linux-arm_64
- linux-x86_64
- mac-universal

For some odd reason the zip file for windows on the official download page is however not named: 
cmake-3.29.2-windows-arm64.zip and the other cmake-3.29.2-linux-aarch64.tar.gz

chrisdutz, 2024-04-26:
https://github.com/cmake-maven-project/cmake-maven-project/issues/56

chrisdutz, 2024-05-17:
I just pushed an update on the documentation on how to setup on windows ... so feel free to have a look. I've tested the build on Arm64 and AMD64/x86_64 Windows and it generally works fine.

chrisdutz, 2024-05-24:
Ok ... changes just merged and should be available now ... updated cmake version to a more recent one supporting all platforms as well as the documentation ... if I don't hear any complaints in the next week or so, I'll treat this issue as done and will close it.

closed 2024-06-21 by chrisdutz