apache/plc4x #1538: [Bug]: plc4j PlcReadRequest excute fail

hyzable, 2024-04-18:
### What happened?

hi:
   PlcReadRequest excute.get() blocked when i connected to a modbus slave simulator software.
   I found that there is no packets got by wireshark, only the tcp handshake packet.
```
 try{
            	logger.info("Read to modbus");
            	PlcReadRequest.Builder mBuilder = mPlcConnection.readRequestBuilder();
            	mBuilder.addTagAddress("coil-1", "coil:1:BOOL[1]");
            	mBuilder.addTagAddress("coil-2", "coil:2:BOOL");
                PlcReadRequest mReadRequest = mBuilder.build();
                PlcReadResponse syncResponse = mReadRequest.execute().get();
                printResponse(syncResponse);
                Thread.sleep(2000);
}
catch (Exception e){
     logger.error("Failed to read from PLC", e);
}
```

The jar used in my project

[image]

### Version

v0.12

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-04-23:
Well if you use that jar you will have a pretty ugly mess of class versions in your classpath as we also use Netty internally and this way Maven will add the netty-all 5 to that. I strongly reccommend not doing that.

If you want us to help you, I think we need a bit more information:
- What type of simulator are you using?
- What's the rest of the code (especially the connection string)?
- Set the log level to TRACE and what does it output?

But you state that nothing is going out ... well ... it might be that your classpath just completely breaks PLC4J because the Modbus drivers would generally always send something out.

hyzable, 2024-04-24:
Thank you for your reply. 
I am a beginner in Java and not very familiar with java developed.
I found that if I create the project as a Maven project through Eclipse and modify the dependency of pom.xml, it can be compiled and run normally. If created as a native Java project and the corresponding jar file is added through running failure information, the issue mentioned will occur. (the get() function blocked notify nothing error only debug info and 4.1.109.Final.jar is only 4k in size on the official website)
Thanks again!

closed 2024-04-24 by hyzable