apache/plc4x #1551: [Feature Request]: support for IEC61850

lucamaf, 2024-04-22:
### What would you like to happen?

As the grid operators are moving towards digitalization, it would be nice to introduce this protocol in the module, to enable scenarios like:
61850 standardization (grid 2 grid messagging)
61850 transformation to MQTT (to gather all devices measurements in a central broker)

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [X] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-11-15:
Do you have access to the spec and devices? Do you want to work on this? If yes, I am sure we will be more than willing to help you.

lucamaf, 2024-12-05:
I don't have access to any specific device, but there are soft implementation of it (based on OpenPLC). You can find the description here: https://www.softxjournal.com/article/S2352-7110(21)00162-X/fulltext

chrisdutz, 2024-12-05:
Yeah ... perhaps someone else will pick this up ... I personally have stopped implementing features others might be liking for free. 8 years of doing that without any significant recognition from the industry I consider enough ... hope you understand. If you want to finance it's development, feel free to ping me.

Just had a look .. the spec is 19000€ (https://www.vde-verlag.de/iec-normen/211712/iec-61850-2024-ser.html) ... I guess I'll pass on that ;-)